esphome/issues #4768: wifi.disable power consumption

qwertz3, 2023-08-04:
### The problem

The power consumption does not meet my expectations when I turn off wifi with "wifi.disable":

- after boot (enable_on_boot: False): ~22mA
- after wifi.enable (power_save_mode: NONE): ~74mA
- after wifi.enable (power_save_mode: HIGH): ~22mA..75mA (mostly below 50mA)
- after wifi.disable: remains ~74mA

My expectation would be that after switching off with "wifi.disable" power consumption would drop to 22mA again.

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
esphome:
  name: d1-mini

esp8266:
  board: d1_mini

logger:
  level: debug
  baud_rate: 0

#api:
#  reboot_timeout: 0s

ota:

wifi:
  enable_on_boot: False
  networks:
    ssid: !secret wifi_ssid
    password: !secret wifi_password
  #power_save_mode: HIGH
  manual_ip:
    static_ip: 192.168.178.104
    gateway: 192.168.178.1
    subnet: 255.255.255.0
    dns1: 192.168.178.1
    dns2: 8.8.8.8

  ap:
    ssid: "${name} Fallback Hotspot"
    password: !secret fallback_password

captive_portal:

web_server:
  port: 80

binary_sensor:
  - platform: gpio
    pin: D1
    id: on_off
    on_click:
      - if:
          condition: wifi.enabled
          then:
            - wifi.disable:
          else:
            - wifi.enable:
```

EternityForest, 2023-08-14:
Seems like this PR https://github.com/esphome/esphome/pull/4916 could be a workaround in the meantime, it doesn't disable WiFi but gets below 22mA

wiltet, 2023-09-17:
I can observe exactly the same behavior on a ESP8285.
In my case, the wifi gets disabled 2min after boot.
The powerconsumption with wifi:off is equal like wifi:on and constantly sending datas.

wifi:disable 10mA (230VAC)
wifi:enable  7,5-10mA (with power_save_mode:LIGHT)
(mesured at the 230VAC mains of a curtain switch SC500W, no actions, only idle)

To proof and be 100% shure about the mesuring I will mesure as well on the 3V3 powerline directly to the chipboard and post results here again.

oddstr13, 2023-09-17:
I can confirm this behavior
[image]

This graph shows current draw over 2 minutes.
Starting powered off (no external power applied), enabling WiFi after 30s, and disabling after another 30s.
without power save enabled gives a very similar graph, just without the dip (marked section) in power usage during enabled WiFi.
```
power_save_mode: HIGH
enable_on_boot: no
```

Board is a Wemos D1 mini, current is measured by powering via the GND+5V pin, input voltage is 5V (supplied by PPK2).

closed 2023-09-20 by jesserockz