esphome/issues #4770: IR receiver after adding on_raw or on_aeha etc........ I do not see log messages (impossible to see IR code)

Jan3726, 2023-08-06:
### The problem

I would like to hear some signal when the IR Receiver is receiving something, i also need to save from the AEHA address in one text sensor and the data in another sensor, this it would be good when need to transmit immediately, without mapping the YAML all the time. Only problem is when i add on_aeha, the logger is stopping to show the received code.

 This is my configuration:

logger:
 level: debug
 logs:
  sensor: none
  text_sensor: none
  switch: none
  binary_sensor: none
  rtttl: none
  script: none
  component: none
  number: none
 baud_rate: 0

and

remote_receiver:
 - pin:
    number: GPIO5
    id: ir_ricevitore
    inverted: true
    mode:
      input: true
      pullup: true
   dump: aeha
   on_aeha:
       then:
        - rtttl.play: 's:d=4,o=7,b=1000:32b4'

### Which version of ESPHome has the issue?

2023.7.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

ssieb, 2023-08-06:
Fixed by https://github.com/esphome/esphome/pull/5124.  It is already in dev and will be in the next release.  You could use it as an external component for now if you want.

closed 2023-08-06 by ssieb

Jan3726, 2023-08-06:
Thank you really much, how i make it so to add the address and data in 2 differents text sensors?

ssieb, 2023-08-06:
I'm not sure what you're asking and this isn't the place for that.  Come ask on the esphome discord server.