esphome/issues #4772: Web UI empty/files not found after HA updates

listlesslife, 2023-08-07:
### The problem

My esphome web ui is blank (picture attached). The add-on logs have errors and are attached.
[image]

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.3

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
[os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 259, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
2023-08-07 00:45:56,167 ERROR 500 GET /devices (0.0.0.0) 7.17ms
2023-08-07 00:47:12,201 ERROR Uncaught exception GET /devices (0.0.0.0)
HTTPServerRequest(protocol='http', host='192.168.0.5:8123', method='GET', uri='/devices', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 145, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 699, in get
    entries = _list_dashboard_entries()
  File "/esphome/esphome/dashboard/dashboard.py", line 608, in _list_dashboard_entries
    files = settings.list_yaml_files()
  File "/esphome/esphome/dashboard/dashboard.py", line 112, in list_yaml_files
    return util.list_yaml_files([self.config_dir])
  File "/esphome/esphome/util.py", line 259, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 259, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
2023-08-07 00:47:12,209 ERROR 500 GET /devices (0.0.0.0) 9.55ms
2023-08-07 00:48:49,157 ERROR Uncaught exception GET /devices (0.0.0.0)
HTTPServerRequest(protocol='http', host='192.168.0.5:8123', method='GET', uri='/devices', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 145, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 699, in get
    entries = _list_dashboard_entries()
  File "/esphome/esphome/dashboard/dashboard.py", line 608, in _list_dashboard_entries
    files = settings.list_yaml_files()
  File "/esphome/esphome/dashboard/dashboard.py", line 112, in list_yaml_files
    return util.list_yaml_files([self.config_dir])
  File "/esphome/esphome/util.py", line 259, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 259, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
2023-08-07 00:48:49,163 ERROR 500 GET /devices (0.0.0.0) 6.77ms
2023-08-07 00:48:55,106 ERROR Uncaught exception GET /devices (0.0.0.0)
HTTPServerRequest(protocol='http', host='192.168.0.5:8123', method='GET', uri='/devices', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 145, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 699, in get
    entries = _list_dashboard_entries()
  File "/esphome/esphome/dashboard/dashboard.py", line 608, in _list_dashboard_entries
    files = settings.list_yaml_files()
  File "/esphome/esphome/dashboard/dashboard.py", line 112, in list_yaml_files
    return util.list_yaml_files([self.config_dir])
  File "/esphome/esphome/util.py", line 259, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 259, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
2023-08-07 00:48:55,109 ERROR 500 GET /devices (0.0.0.0) 3.51ms
2023-08-07 00:49:38,314 ERROR Uncaught exception GET /devices (0.0.0.0)
HTTPServerRequest(protocol='http', host='127.0.0.1:64447', method='GET', uri='/devices', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 145, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 699, in get
    entries = _list_dashboard_entries()
  File "/esphome/esphome/dashboard/dashboard.py", line 608, in _list_dashboard_entries
    files = settings.list_yaml_files()
  File "/esphome/esphome/dashboard/dashboard.py", line 112, in list_yaml_files
    return util.list_yaml_files([self.config_dir])
  File "/esphome/esphome/util.py", line 259, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 259, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
2023-08-07 00:49:38,317 ERROR 500 GET /devices (0.0.0.0) 4.22ms
2023-08-07 00:50:07,571 ERROR Uncaught exception GET /devices (0.0.0.0)
HTTPServerRequest(protocol='http', host='192.168.0.5:8123', method='GET', uri='/devices', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 145, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 699, in get
    entries = _list_dashboard_entries()
  File "/esphome/esphome/dashboard/dashboard.py", line 608, in _list_dashboard_entries
    files = settings.list_yaml_files()
  File "/esphome/esphome/dashboard/dashboard.py", line 112, in list_yaml_files
    return util.list_yaml_files([self.config_dir])
  File "/esphome/esphome/util.py", line 259, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 259, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
2023-08-07 00:50:07,575 ERROR 500 GET /devices (0.0.0.0) 4.91ms
```

### Additional information

Updated HA OS to 10.4 and Core to 2023.8.1 from 2023.7.3.  That gave me all sorts of problems (not just the mqtt issue), so I rolled back to 2023.7.3. Now my web ui is blank and the add-on is throwing errors. I did not check if this was a problem with 2023.8.1 because 2023.8.1 was giving too many errors for my comfort.

brilvio, 2023-09-28:
I have the same problem

boardyuk, 2023-09-28:
I'm also getting this on a brand new 2023.9.1 add-on.

`Home Assistant 2023.9.3
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest`

`2023-09-28 21:37:55,438 INFO 200 GET /static/js/esphome/monaco-editor/esm/vs/editor/editor.worker.js (0.0.0.0) 0.54ms
2023-09-28 21:37:55,582 INFO 101 GET /ace (0.0.0.0) 1.45ms
2023-09-28 21:37:55,623 INFO Running command 'esphome --dashboard -q vscode --ace /config/esphome'
2023-09-28 21:38:04,779 ERROR Uncaught exception POST /edit?configuration=secrets.yaml (0.0.0.0)
HTTPServerRequest(protocol='http', host='home.sanitised.com', method='POST', uri='/edit?configuration=secrets.yaml', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 151, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 176, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 1037, in post
    with open(file=settings.rel_path(configuration), mode="wb") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome/secrets.yaml'
2023-09-28 21:38:04,781 ERROR 500 POST /edit?configuration=secrets.yaml (0.0.0.0) 3.83ms
2023-09-28 21:38:06,004 ERROR Uncaught exception POST /edit?configuration=secrets.yaml (0.0.0.0)
HTTPServerRequest(protocol='http', host='home.santised.com', method='POST', uri='/edit?configuration=secrets.yaml', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 151, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 176, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 1037, in post
    with open(file=settings.rel_path(configuration), mode="wb") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome/secrets.yaml'
2023-09-28 21:38:06,006 ERROR 500 POST /edit?configuration=secrets.yaml (0.0.0.0) 2.99ms
2023-09-28 21:38:10,167 ERROR Uncaught exception GET /devices (0.0.0.0)
HTTPServerRequest(protocol='http', host='home.sanitised.com', method='GET', uri='/devices', version='HTTP/1.1', remote_ip='0.0.0.0')
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/tornado/web.py", line 1784, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 151, in decorator
    return func(self, *args, **kwargs)
  File "/esphome/esphome/dashboard/dashboard.py", line 743, in get
    entries = _list_dashboard_entries()
  File "/esphome/esphome/dashboard/dashboard.py", line 654, in _list_dashboard_entries
    files = settings.list_yaml_files()
  File "/esphome/esphome/dashboard/dashboard.py", line 118, in list_yaml_files
    return util.list_yaml_files([self.config_dir])
  File "/esphome/esphome/util.py", line 265, in list_yaml_files
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
  File "/esphome/esphome/util.py", line 265, in <listcomp>
    [os.path.join(folder, p) for folder in folders for p in os.listdir(folder)]
FileNotFoundError: [Errno 2] No such file or directory: '/config/esphome'
2023-09-28 21:38:10,167 ERROR 500 GET /devices (0.0.0.0) 0.68ms
2023-09-28 21:38:11,249 INFO 200 GET /edit?configuration=secrets.yaml (0.0.0.0) 1.46ms
2023-09-28 21:38:11,400 INFO 101 GET /ace (0.0.0.0) 1.09ms
2023-09-28 21:38:11,450 INFO Running command 'esphome --dashboard -q vscode --ace /config/esphome`