esphome/issues #4774: Sprinkler reset_resume() function not recognized

dgarciadonado, 2023-08-07:
### The problem

It seems that the reset_resume() function cannot be used, it is flagged as Public but for some reason I cannot use. (I think I am calling it properly)
This function is very useful to completely clear the paused valves.

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.2

### What platform are you using?

ESP32

### Board

Shelly 2.5 plus

### Component causing the issue

Sprinkler

### Example YAML snippet

```yaml
- platform: template
    name: ${upper_devicename} stop_all_valves
    id: stop_all_valves
    on_press:
      then:   
        - sprinkler.shutdown: main_sprinkler       
        - sprinkler.reset_resume: main_sprinkler
```

### Anything in the logs that might be useful for us?

```txt
Failed config

button.template: [source /config/esphome/sprinkler/general_modes.yaml:59]
  platform: template
  name: shelly_2_5_plus_riego stop_all_valves
  id: stop_all_valves
  on_press: 
    then: 
      - sprinkler.shutdown: main_sprinkler
      - 
        Unable to find action with the name 'sprinkler.reset_resume'.
        sprinkler.reset_resume: main_sprinkler
```

ssieb, 2023-08-07:
It's a public C++ method, but there's no action for it.  You would have to call that from a lambda.

closed 2023-08-07 by ssieb

dgarciadonado, 2023-08-07:
Thanks for your help!!
I let here the code just for reference:

button:
  # Close all cycles
  - platform: template
    name: ${upper_devicename} stop_all_valves
    id: stop_all_valves
    on_press:
      then:   
        - sprinkler.shutdown: main_sprinkler       
        - lambda: |-
            id(main_sprinkler).reset_resume();

ssieb, 2023-08-07:
Btw, you don't need the `|-` for a single line.  Just use `- lambda: id(main_sprinkler).reset_resume();`