esphome/issues #4776: BT Proxy not working

ddppddpp, 2023-08-08:
### The problem

I've installed BT Proxy on a M5Stack Atom Lite using the [Web Installer](https://esphome.github.io/bluetooth-proxies/) and added it to HA but it didn't detect any device, although there should be a couple in the root the controller is sitting in (A RPi3B+, a BT speaker, a Garmin watch, etc.)
I've manually added logging to the configuration on Debug and then Verbose level to understand what's wrong.
I noticed it takes ~5mins until the 'boot seems succesful ' and 'starting scan' messages appear, not sure if this is normal or not.
On Verbose, I can see only this couple of lines being repeated
```
[12:28:49][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[12:28:49][V][bluetooth_proxy:058]: Proxying 1 packets
[12:28:50][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[12:28:50][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[12:28:50][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[12:28:50][V][bluetooth_proxy:058]: Proxying 3 packets
[12:28:51][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[12:28:51][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[12:28:51][V][bluetooth_proxy:058]: Proxying 2 packets
[12:28:51][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[12:28:51][V][bluetooth_proxy:058]: Proxying 1 packets
```

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.1

### What platform are you using?

ESP32

### Board

M5Stack Atom Lite

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-e9c454
  friendly_name: m5-stack-lite-bt-proxy
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: asdfasdf

logger:
  level: DEBUG

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

alfredocdmiranda, 2023-08-19:
I have the same issue. BT Proxy says that it has proxied the message, however, I can't see anything in HA.

`substitutions:
  name: esp32-bluetooth-proxy
  friendly_name: Bluetooth Proxy

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: true
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: !secret
  password: !secret

  ap:
    ssid: "Bluetooth-Proxy Fallback Hotspot"
    password: !secret

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/esp32-generic.yaml@main

esp32_ble_tracker:
  scan_parameters:
    active: true

bluetooth_proxy:
  active: true
`

matthew73210, 2023-08-27:
I also have the same issue. After poking around on the forum, what I've found out is that if the device can't be tracked on ha itself. i.e can't be added. Then it won't do anything, or as far as i can tell (got told off haha). If true it's misleading information on the website that puts people off.

https://community.home-assistant.io/t/atom-m5-stack-ble-proxy-detection-problems/603921/4

alfredocdmiranda, 2023-08-27:
In the end the BT proxy was working fine, my problem was related to the firmware I was using in my Xiaomi Thermometers that were not being identified by Home Assistant. After changing the firmware and some configurations, everything is working fine.

matthew73210, 2023-08-27:
> In the end the BT proxy was working fine, my problem was related to the firmware I was using in my Xiaomi Thermometers that were not being identified by Home Assistant. After changing the firmware and some configurations, everything is working fine.

Glad you got it working. If I may ask, what did you change? Bit bummed that you can't just track all ble devices but hey, such is life.

alfredocdmiranda, 2023-08-27:
First, I was using atc1441's firmware. Then, I had to use the most recent [pvvx's version](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) and change the Advertisement Type to BTHome. After that all my thermometers started working fine.