esphome/issues #4777: On 2023.7.1, Initial Display Config with 'greyscale: True' Causes Device Initialize to Seemingly Fail

oaa8, 2023-08-10:
### The problem

After upgrading to 2023.7.1, my Inkplate6 plus would no longer finish booting up nor connect to show logs remotely.  I narrowed things down to enabling the greyscale mode in the display config.  Here's the configuration:
```YAML
display:
- platform: inkplate6
  id: inkplate_display
  greyscale: True # Replacing this with greyscale: false allows for proper booting
  model: inkplate_6_plus
  partial_updating: false
  update_interval: 60000s
```

Currently my work around is to have `greyscale: false` and then wait until after boot up to then switch greyscale back on with `id(inkplate_display).set_greyscale(true);`

In case it's helpful, this is the workaround I'm using in case it helps anyone else:

```YAML
esphome:
  on_boot:
    priority : -100.0
    then:
      - wait_until: api.connected
      - lambda: id(inkplate_display).set_greyscale(true);
      - delay: 5s
      - component.update: inkplate_display
```

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.08.1

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

display

### Example YAML snippet

```yaml
display:
- platform: inkplate6
  id: inkplate_display
  greyscale: True
  model: inkplate_6_plus
  partial_updating: false
  update_interval: 60000s

  ckv_pin: 32
  sph_pin: 33
  gmod_pin:
    mcp23xxx: mcp23017_hub
    number: 1
  gpio0_enable_pin:
    mcp23xxx: mcp23017_hub
    number: 8
  oe_pin:
    mcp23xxx: mcp23017_hub
    number: 0
  spv_pin:
    mcp23xxx: mcp23017_hub
    number: 2
  powerup_pin:
    mcp23xxx: mcp23017_hub
    number: 4
  wakeup_pin:
    mcp23xxx: mcp23017_hub
    number: 3
  vcom_pin:
    mcp23xxx: mcp23017_hub
    number: 5
```

### Anything in the logs that might be useful for us?

```txt
Unfortunately the device doesn't connect to the network while in this state and I'm not sure how to acquire logs apart from through the network (sorry I'm new to ESPHome)
```

ssieb, 2023-08-10:
You can get serial logs by plugging it into your computer and using either the web installer or a program like putty.

oaa8, 2023-08-10:
Thank you very much!  Here are logs from one of the failed boots in the boot loop before going into safe mode:

```
ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:262]: Log initialized
[C][ota:469]: There have been 8 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:183]: Performing I2C bus recovery
[C][mcp23017:010]: Setting up MCP23017...
[C][power_supply:010]: Setting up Power Supply...
[C][switch.gpio:011]: Setting up GPIO Switch 'battery_read_mosfet'...
[D][switch:016]: 'battery_read_mosfet' Turning OFF.
[D][switch:055]: 'battery_read_mosfet': Sending state OFF
[D][switch:016]: 'battery_read_mosfet' Turning OFF.
[C][switch.gpio:011]: Setting up GPIO Switch 'Inkplate Touchscreen Enabled'...
[D][switch:012]: 'Inkplate Touchscreen Enabled' Turning ON.
[D][switch:055]: 'Inkplate Touchscreen Enabled': Sending state ON
[D][switch:012]: 'Inkplate Touchscreen Enabled' Turning ON.
[C][light:035]: Setting up light 'Inkplate Backlight'...
[D][light:036]: 'Inkplate Backlight' Setting:
[D][light:041]:   Color mode: 
[D][light:085]:   Transition length: 0.2s
[D][template.switch:046]:   Restored state OFF
[D][switch:016]: 'Inkplate Greyscale mode' Turning OFF.
[D][template.switch:046]:   Restored state OFF
[D][switch:016]: 'Inkplate Partial Updating' Turning OFF.
[C][ektf2232:021]: Setting up EKT2232 Touchscreen...
[D][binary_sensor:034]: 'Inkplate Status': Sending initial state OFF
CORRUPT HEAP: Bad head at 0x3f83b340. Expected 0xabba1234 got 0x00000000

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x40083dc5:0x3ffb24300x4008dd0d:0x3ffb2450 0x400933bd:0x3ffb2470 0x40093003:0x3ffb25a0 0x40084229:0x3ffb25c0 0x400933ed:0x3ffb25e0 0x400db04a:0x3ffb2600 0x400db20e:0x3ffb2640 0x400e5316:0x3ffb2660 0x40174185:0x3ffb26b0 0x400e4eb8:0x3ffb26d0 0x400e825e:0x3ffb2700 0x400ef53e:0x3ffb2820 
```

closed 2024-01-16 by jesserockz