esphome/issues #4779: ESPHome ESP32 find no DS18B Sensor anymore

shgmongo, 2023-08-10:
### The problem

My ESPHome (ESP32) was working fine with 9 DS18B20 Sensors for 12 hr. Then the connnection failed and the ESPHome find no Sensors anymore.
Firmware of ESPHome ist 2023.7.1
I don´t know if there was an automatic Firmeware update during the night.

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: esphome-03
  friendly_name: esphome-03

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "vL0ZIoJiF/goYE2u92ovGALEAj"

ota:
  password: "2c30651f4d9c4f8d"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Esphome-03 Fallback Hotspot"
    password: "PnIzL8Z0grw8"

captive_portal:

esp32_ble_tracker:

dallas:
  pin:
    number: GPIO23
    #mode:
     # input: true
     # pullup: true
  update_interval: 300s 

sensor:
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:2C:C4:10"
    temperature:
      name: "Thaus unten Temp"
    humidity:
      name: "Thaus unten Hum"
    battery_level:
      name: "Thaus unten  BaLev"
    battery_voltage:
      name: "Thaus unten  BaVo"
    
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:DF:E4:21"
    temperature:
      name: "HeizRaum Temp"
    humidity:
      name: "HeizRaum  Hum"
    battery_level:
      name: "HeizRaum  BaLev"
    battery_voltage:
      name: "HeizRaum  BaVo"

#Dallas
  - platform: dallas
    address: 0x57d64e421f64ff28
    name: "TempSensor-01"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0xfc3077421f64ff28
    name: "TempSensor-02"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0x43566a9c1e64ff28
    name: "TempSensor-03"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0x2cff4f421f64ff28
    name: "TempSensor-04"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0x0fef93431f64ff28
    name: "TempSensor-05"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0xc0e1aa411f64ff28
    name: "TempSensor-06"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0xdee9a4411f64ff28
    name: "TempSensor-07"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0xdc26a9411f64ff28
    name: "TempSensor-08"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1

  - platform: dallas
    address: 0xf85b7c9c1e64ff28
    name: "TempSensor-09"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 1
```

shgmongo, 2023-08-10:
[image]

HassioAwi, 2023-08-18:
https://github.com/esphome/issues/issues/3909

closed 2024-01-26 by github-actions[bot]