esphome/issues #4780: Midea_AC

Daniel210117, 2023-08-10:
### The problem

with the rest api i can't switch fan_mode or swing_mode.
All other work fine.

Example:
http://192.168.13.126/climate/midea_climate/set?mode=off
with this "post" I can switch off my AC.

http://192.168.13.126/climate/midea_climate/set?swing_mode=off
this don't work, but I see the correct swing mode with "get" http://192.168.13.126/climate/midea_climate

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
esphome:
  name: air-conditioner
  platform: ESP8266
  board: esp12e
  name_add_mac_suffix: true

wifi:

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "AC-wifi"
    password: "slwf01pro"

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  
  
climate:
  - platform: midea
    name: Midea Climate         # Use a unique name.
    period: 1s                  # Optional
    timeout: 2s                 # Optional
    num_attempts: 3             # Optional
    autoconf: true              # Autoconfigure most options.
    beeper: true                # Beep on commands.
    visual:                     # Optional. Example of visual settings override.
      min_temperature: 17 °C    # min: 17
      max_temperature: 30 °C    # max: 30
      temperature_step: 0.5 °C  # min: 0.5
    supported_modes:            # All capabilities in this section detected by autoconf.
      - FAN_ONLY                # This capability is always used.
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
    custom_fan_modes:
      - SILENT
      - TURBO
    supported_presets:          # All capabilities in this section detected by autoconf.
      - ECO
      - BOOST
      - SLEEP                   # This capability is always used.
    custom_presets:             # All capabilities in this section detected by autoconf.
      - FREEZE_PROTECTION
    supported_swing_modes:
      - VERTICAL                # This capability is always used.
      - HORIZONTAL
      - BOTH
    outdoor_temperature:        # Optional. Outdoor temperature sensor (may display incorrect values after long inactivity).
      name: Temp
    power_usage:                # Optional. Power usage sensor (only for devices that support this feature).
      name: Power
    humidity_setpoint:          # Optional. Indoor humidity sensor (only for devices that support this feature).
      name: Humidity

web_server:
  port: 80
```

Tarik2142, 2023-10-09:
seems like this API missing in web_server.cpp
[image]

jens-schiffke, 2023-10-14:
I added the missing modes.
Swing_mode and fan_mode now work.
Unfortunately the (custom_)preset_modes do not work.

[image]

closed 2024-09-22 by jesserockz