esphome/issues #4781: Font conversion stopped working after updating to ESPHome 2023.8.0b1

srusinov, 2023-08-11:
### The problem

I get the following error when recompiling my configuration for esp32 and ttgo display:

INFO ESPHome 2023.8.0b1
INFO Reading configuration /config/esphome/esphome-web-air.yaml...
INFO Detected timezone 'Europe/Sofia'
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
ERROR Could not load truetype file /config/esphome/arial.ttf: cannot import name '_imagingft' from 'PIL' (/usr/local/lib/python3.9/dist-packages/PIL/__init__.py)

before updating to ESPHome 2023.8.0b1 there was no issues with the same configuration

I believe this is caused by this change: 
https://github.com/esphome/esphome/pull/5074
 

### Which version of ESPHome has the issue?

ESPHome 2023.8.0b1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.8.1 Supervisor 2023.08.1 Operating System 10.4 Frontend 20230802.0 - latest

### What platform are you using?

ESP32

### Board

ODROID-XU4

### Component causing the issue

font:

### Example YAML snippet

```yaml
esphome:
  name: "esphome-web-air"
  friendly_name: ESPHome Web c7f3b8

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXX"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-C7F3B8"
    password: "XXXX"

captive_portal:

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO19

time:
  - platform: homeassistant
    id: esptime

display:
  - platform: st7789v
    model: TTGO TDisplay 135x240
    backlight_pin: GPIO4
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO23
    update_interval: 1s
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");
      it.image(0, 50, id(my_image));
      it.strftime(0, 160, id(font1), id(my_gray), TextAlign::LEFT, "%H:%M:%S", id(esptime).now());

font:
  - file: "arial.ttf"
    id: font1
    size: 20

image:
  - file: "leaf.png"
    id: my_image
    resize: 100x100
    type: RGB24

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%
```

### Anything in the logs that might be useful for us?

```txt
ERROR Could not load truetype file /config/esphome/arial.ttf: cannot import name '_imagingft' from 'PIL' (/usr/local/lib/python3.9/dist-packages/PIL/__init__.py)
```

ssieb, 2023-08-11:
Try re-installing the addon.

srusinov, 2023-08-11:
Thanks, installing the non-betta version fixed the issue. Still, I believe there is something wrong with ESPHome 2023.8.0b1 and https://github.com/esphome/esphome/pull/5074

randybb, 2023-08-11:
there is not. your config is working with the current stable, beta and dev version.

srusinov, 2023-08-11:
Ok, then what is the reason for this error? Somehow my ESPHome toolchains messed up after the upgrade to 8.0b1?

pierlon, 2023-08-16:
Got the same error on ESPHome 2023.8.0b3. It seems the `libfreetype-dev` apt package needs to be added to the Docker image.

closed 2023-08-17 by jesserockz