esphome/issues #4784: ESP32 with Arduino gets no IPv6

Nardol, 2023-08-13:
### The problem

I have set a speaker for Hassist with a m5stack atom-echo and enabled IPv6 by setting `enable_ipv6` to `true` in the `network` section.
But after uploading it, no IPv6 is set at all.
First of all I thought it was only MDNS which did not resolve using IPv6 at all but in fact it looks like there is no IPv6 at all.

### Which version of ESPHome has the issue?

2023.8.0b1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

m5stack-atom-echo

### Component causing the issue

network

### Example YAML snippet

```yaml
esphome:
  name: enceinte-test
  friendly_name: "Enceinte test"
  project:
    name: m5stack.atom-echo
    version: "1.0"
  min_version: 2023.5.0

esp32:
  board: m5stack-atom
  framework:
    type: arduino

dashboard_import:
  package_import_url: github://esphome/media-players/m5stack-atom-echo.yaml@main

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key
  fast_connect: true
  manual_ip:
    static_ip: "${ip}"
    gateway: "192.168.xxx.xxx"
    subnet: "255.255.255.0"
    dns1: "8.8.8.8"
    dns2: "8.8.4.4"
  ap:
    password: !secret ap_key

network:
  enable_ipv6: true
```

### Additional information

Also tested by @jeremyp3 with an AZ-Delivery devkit with the same result.
With ESP-IDF IPv6 is set but not with Arduino.

Nardol, 2023-08-13:
As a note, IPv6 are assigned using SLAAC and DNS by RDNSS.
The only issue with ESP-IDF is the known #3896 but with Arduino no IPv6 at all, no fe80.

randybb, 2023-08-13:
https://esphome.io/components/network it is available only on esp-idf

Nardol, 2023-08-13:
@randybb I don't read the same in the beta changelog.

randybb, 2023-08-13:
oh beta. then let's see.

HeMan, 2023-08-15:
I just tested and I got IPv6 addresses. Could you add
```
logger:
  level: VERY_VERBOSE
```
And upload output.

Nardol, 2023-08-15:
Also tested again with 2023.8.0b2 with same result.

About `VERY_VERBOSE` the problem is I only get IPv6 address if I set log level higher than `DEBUG`... Noticed when testing as you asked.

HeMan, 2023-08-15:
> Also tested again with 2023.8.0b2 with same result.
> 
> About `VERY_VERBOSE` the problem is I only get IPv6 address if I set log level higher than `DEBUG`... Noticed when testing as you asked.

Oh, that's interresting!

closed 2023-08-17 by jesserockz