esphome/issues #4789: Default ("main") device entity in MQTT discovery

dotdoom, 2023-08-14:
### The problem

Default / main entity for a device is an entity which name is `null`:
https://www.home-assistant.io/integrations/mqtt/#naming-of-mqtt-entities

esphome currently doesn't allow such entities, by forbidding `null` as a value for `name` attribute.  It allows `None` instead:
https://esphome.io/components/sensor/index.html#base-sensor-configuration

But that has a different effect: `name` populated to MQTT broker is the `friendly_name` of the device, not `null`.  So if my device `friendly_name` is `"Bathroom heater"` and I have a climate entity with `name: None`, the name sent to HA will be `"Bathroom heater"`.

Now HA is changing `friendly_name` approach to always prefix device name to the entity's `friendly_name` ([blog entry](https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities), [PSA](https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099)).  As a result, the entity name will become `"Bathroom heater Bathroom heater"` (or actually `"bathroomheater Bathroom heater"` because of #4557).  Such naming is obviously redundant.

Are there any plans to solve this redundancy problem before the "breaking" release of HA 2024.02?  If not yet, would you consider a PR to change that or am I doing it wrong?

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP8266

### Board

nodemcu

### Example YAML snippet

```yaml
# esphome config
substitutions:
  room: "Bathroom"

esphome:
  # https://esphome.io/components/esphome.html#configuration-variables
  # May only contain lowercase, digits and hyphens, up to 24chars.
  name: heater
  # If you have more than one such device per network & HA deployment.
  name_add_mac_suffix: true
  friendly_name: "${room} heater"

climate:
  # https://esphome.io/components/sensor/index.html#base-sensor-configuration
  # We can use "None" to indicate that it's the main entity for the device.
  # This is the only place where "friendly_name" is used, AFAICT. It's used by
  # esphome itself, and will be replaced by friendly_name from section above.
  # Setting to "null" is not allowed by esphome configuration check.
  - name: None
    platform: xxx

switch:
  - name: "status led"
    platform: yyy
```

### Anything in the logs that might be useful for us?

```txt
Setting "name: null" results in:

Expected boolean value, but cannot convert None to a boolean. Please use 'true' or 'false'.
```

mtl010957, 2023-12-13:
I think this should be fixed by PR #[5897](https://github.com/esphome/esphome/pull/5897)

dotdoom, 2023-12-23:
It is!  Verified with 2013.12.3, `"name": ""` is present in entity configs topic of MQTT.

closed 2023-12-23 by dotdoom