esphome/issues #4790: Graph stops updating when value is 0

hapklaar, 2023-08-14:
### The problem

I'm graphing the amount of power that I'm returning to the grid. When this drops to 0, the graph stops updating and seems to freeze.

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP32

### Board

LilyGO TTGO T-Display-S3

### Example YAML snippet

```yaml
sensor:
  - platform: homeassistant
    id: dsmr_power_returned
    #entity_id: sensor.dsmr_power_returned
    entity_id: sensor.dsmr_produced_in_watts
    unit_of_measurement: 'W'
    force_update: true

graph:
  - id: dsmr_power_graph
    duration: 60min
    x_grid: 10min
    y_grid: 500
    width: 318
    height: 63
    border: false
    traces:
      - sensor: dsmr_power_returned
        line_type: SOLID
        line_thickness: 2
        color: color_green

display:
  - platform: tdisplays3
    rotation: 270
    update_interval: 2s
    lambda: |-
      ....
      it.graph(1, 107, id(dsmr_power_graph), color_teal_blue);
```

### Additional information

Already added `force_update: true` to the sensor config, but this made no difference

ssieb, 2023-08-14:
I assume that HA isn't sending duplicate values.  The force_update is for pushing duplicate data **to** HA, so has no effect here.  This isn't an esphome issue.

closed 2023-08-14 by ssieb

hapklaar, 2023-08-14:
HA is actually sending updates for this sensor, it is updated every 10 seconds. Also when value is 0.

[image]

ssieb, 2023-08-15:
The graph is time based, but only updates when there's a new value.  Can you paste the log showing that the sensor is getting updates?  If you change the log level to verbose, the graph will also log when it gets a new value.

hapklaar, 2023-08-15:
After setting the logger to verbose, I'm not seeing updates from HA for this sensor when the value is 0. 

- Does this mean the HA api is not providing esphome with any values for this sensor even though I see it being updated in HA (with value 0) every 10 seconds? Can we somehow force this?
- As the graph trace stops when it's not receiving any values, it gets out of sync with the other traces. Is there a way to keep all traces remain in sync even when one of them is not receiving updates because of above behavior?

It seems like currently it is not possible to graph a (temporary) non-changing value, or am I misunderstanding?

ssieb, 2023-08-15:
Yes, HA isn't sending the duplicates.  There's no way to force it from the device side, you could maybe open an issue on HA for the esphome integration.  You could use the `heartbeat:` filter to ensure that you get regular updates.
What other traces?  Your graph config only has one trace.
It's not about the value not changing, it's about the sensor not updating.

hapklaar, 2023-08-15:
I removed the other trace temporarily for clarity. This is my actual code with 2 traces:

```yaml
graph:
  - id: dsmr_power_graph
    duration: 60min
    x_grid: 10min
    y_grid: 500
    width: 318
    height: 63
    border: false
    traces:
      - sensor: dsmr_power_delivered
        line_type: SOLID
        line_thickness: 3
        color: color_red
      - sensor: dsmr_power_production
        line_type: SOLID
        line_thickness: 2
        color: color_green
```

The green trace will halt when the sensor hits 0, the red one will continue as there is always some power delivery going on. This means I will keep the last 60 minutes of green trace until it rises up from 0 again the next day.

Wouldn't a better solution be if the trace that doesn't get new data would just continue with the last value it had, keeping them in sync?

ps thanks for taking the time to work with me. I hope I was able to present the issue, English is not my primary language ;)

ssieb, 2023-08-15:
It's complicated.  The graph data only gets updated when data arrives.  I think the current method is wrong in that it fills in the time since the last sample with the new data, when it should actually be the old data.  File a new issue that the trace doesn't update when the sensor doesn't update and include that last line from me.  This issue has nothing to do with the value.
A workaround, as I mentioned, is to use the `heartbeat` filter on the sensor.  See the docs for how to use it.

hapklaar, 2023-08-16:
`heartbeatfilter` did the trick for me, thanks for the tip