esphome/issues #4791: On_release while triggers unexpectedly

cmatte, 2023-08-14:
### The problem

Hi,

Intent here is to set-up an on-board light switch automation driven by a motion sensor, switching off the light 5m after the motion ends. The same automation in home assistant works perfectly, yet the on-board version is required as it would also operate without Internet connection to the home assistant server.

The automation documentation guides through a few different approaches to achieve this, however after applying the most clean one, the behavior is wildly unexpected, or well, maybe the interpretation of the documentation is not going too well, in which case any support is welcome 😊

Code, expected behavior, and actual behavior, below.

Thanks in advance!

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP32-IDF

### Component causing the issue

automations

### Example YAML snippet

```yaml
light:
  - platform: binary
    name: "LED Cucina"
    output: output_led_cucina
    id: led_cucina

output:
  - id: output_led_cucina
    platform: gpio
    pin: GPIO16

binary_sensor:
  - platform: gpio
    pin: GPIO13
    name: "Movimento Piano Cucina"
    id: movimento_piano_cucina
    device_class: motion
    on_press:
      then:
      - light.turn_on: led_cucina
    on_release:
      - while:
          condition:
            binary_sensor.is_off: movimento_piano_cucina
          then:
          - logger.log: "Still executing"
          - delay: 300s
          - light.turn_off: led_cucina
```

### Anything in the logs that might be useful for us?

```txt
[20:42:30][C][gpio.output:010]: GPIO Binary Output:
[20:42:30][C][gpio.output:011]:   Pin: GPIO16
[20:42:30][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Movimento Piano Cucina'
[20:42:30][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[20:42:30][C][gpio.binary_sensor:016]:   Pin: GPIO13
[20:42:30][C][light:103]: Light 'LED Cucina'
[20:42:30][C][captive_portal:088]: Captive Portal:
[20:42:31][C][mdns:112]: mDNS:
[20:42:31][C][mdns:113]:   Hostname: cucina-esp32
[20:42:31][C][ota:093]: Over-The-Air Updates:
[20:42:31][C][ota:094]:   Address: 192.168.4.32:3232
[20:42:31][C][ota:097]:   Using Password.
[20:42:31][C][api:138]: API Server:
[20:42:31][C][api:139]:   Address: 192.168.4.32:6053
[20:42:31][C][api:141]:   Using noise encryption: YES
[20:42:35][D][binary_sensor:036]: 'Movimento Piano Cucina': Sending state OFF
[20:42:35][D][main:293]: Still executing
[20:42:37][D][light:036]: 'LED Cucina' Setting:
[20:42:37][D][light:047]:   State: OFF
[20:42:37][D][main:293]: Still executing
[20:42:44][D][light:036]: 'LED Cucina' Setting:
[20:42:44][D][main:293]: Still executing
[20:42:54][D][binary_sensor:036]: 'Movimento Piano Cucina': Sending state ON
[20:42:54][D][light:036]: 'LED Cucina' Setting:
[20:42:54][D][light:047]:   State: ON
[20:42:57][D][binary_sensor:036]: 'Movimento Piano Cucina': Sending state OFF
[20:42:57][D][main:293]: Still executing
```

### Additional information

Expectation is that the while drops running the delay action when the sensor goes on, as the while condition to keep running is that the motion sensor keeps off, instead, the logs and behavior confirm that the on_release while actions keep running regardless there has been a new on_press, which should be, instead, blocking the previous waiting time, and only re-trigger the on_release with its consequent interruption of the motion. The behavior is that the led switches off unexpectedly.

ssieb, 2023-08-14:
github issues are not for support requests.  Come ask on the esphome discord or one of the forums.  Look at the delay filters.  I expect that's what you're wanting.

ssieb, 2023-08-14:
Everything is working exactly as you've told it to.  I'm not entirely clear on what you think it should be doing, but it's not a bug.  As I said, if you still need help, then come visit the discord server.

cmatte, 2023-08-14:
What's expected is explained above, it is to interrupt the while cycle when the condition turns false as it should do any time the motion sensor goes ON. Instead, it keeps running regardless, the documentation confirms the while actions run until the condition is false, to me that clearly is not happening, therefore the issue raised here. Nonetheless, I will reach out to the discord channel to understand more...