esphome/issues #4792: MQTT subscribe causes reboot

mrkvo, 2023-08-14:
### The problem

Hello! 
 I am controlling digital to analog converter via MQTT subscribe and since AGES I have problem:

When I am frequently sending to MQTT some data (once per second, its just integer value from 0-1023), and its subscribed by ESPHome, its reboots spontainously 1-2 times per day.

### Which version of ESPHome has the issue?

all versions since 2021 - 2023

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP32

### Board

My custom

### Component causing the issue

mqtt_subscribe

### Example YAML snippet

```yaml
text_sensor:
  #########################
  # Current over mqtt
  #########################   
  - platform: mqtt_subscribe
    name: "CurrentDACMQTT"
    id: CurrentDACMQTT
    topic:  ${devicename_lower}/service/current_dac
    qos: 0
```

### Anything in the logs that might be useful for us?

```txt
1, tag json and message  [1;31m[E][json:041]: Could not allocate memory for JSON document! Requested 512 bytes, largest free heap block: 1524 bytes
```

### Additional information

Sometimes it says that error message, but I dont know if its due to this.

VNRARA, 2023-12-08:
Same issue here. Random reboots, I'm going to check if mqtt subscribe or the logger is the problem. Maybe the values are to big to handle and aren't being truncated?

spuder, 2024-10-11:
Do you have `api` disabled? The docs mention an incompatibility between mqtt and api

https://esphome.io/components/mqtt.html

VNRARA, 2024-10-11:
Yes