esphome/issues #4794: Deep sleep does not shut down connection with HA gracefuly

Jordan-Szwejda, 2023-08-15:
### The problem

That's follow up a problem that sensor value of a device which reports voltage every 1h and goes deep sleep is reported as Unavailable in home assistant.
The original thread was created here:
https://github.com/home-assistant/core/issues/98221#issuecomment-1677961214

The node should close correctly API connection to HA. Instead of this what can be observed in logs is:

`2023-08-15 02:54:27.493 DEBUG (MainThread) [homeassistant.components.esphome.manager] carvoltage-esp: 192.168.0.208 disconnected (expected=False), running disconnected callbacks`.

It is possible that the node enters deep sleep too early and communication with HA is not finished correctly. The node has weak wifi signal but there is not problem neither with transmitting sensor values nor with doing OTA updates so I can assume that connection is reliable enough to perform such a connection closing procedure.

I am attaching logs from HA:

[ha-logs.txt](https://github.com/esphome/issues/files/12342426/ha-logs.txt)

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.1

### What platform are you using?

ESP8266

### Board

esp01_1m

### Example YAML snippet

```yaml
esphome:
  name: carvoltage-esp
  friendly_name: carvoltage-esp

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "gRoLBS0F6kxfXi6tqO8ye3WUeA+Ny70C6Y35HbyKTi0="

ota:
  password: "e54703412158e61bfff9ced3895c05a6"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  id: wifi_id
  manual_ip:
    static_ip: 192.168.0.208
    gateway: 192.168.0.1
    subnet: 255.255.255.0

deep_sleep:
  id: deep_sleep_id
  sleep_duration: 60s
   
sensor:
  - platform: adc
    id: adc_sensor
    pin: A0
    name: "Car Voltage"
    update_interval: 60000s     
    filters:
     - multiply: 14.02
  - platform: homeassistant
    name: "Deep Sleep Duration"
    id: deep_sleep_time
    entity_id: input_number.carvoltage_deep_sleep_time
    on_value:
      then:
        - logger.log:
            format: "Received deep sleep duration"
        - lambda: 'id(adc_sensor).update();'  
        - if:
            condition: 
              lambda: 'return id(deep_sleep_time).state>0;' 
            then:
              - deep_sleep.enter:
                  id: deep_sleep_id
                  sleep_duration: !lambda |-
                    return id(deep_sleep_time).state * 1000;
  - platform: uptime
    id: uptime_sensor
    name: Uptime Sensor
    internal: True
    on_value:
      then:
        - if:
            condition:            
              lambda: 'return id(uptime_sensor).state>=180 && !id(wifi_id).is_connected();'
            then:
              - deep_sleep.enter:
                  id: deep_sleep_id
                  sleep_duration: 3600s
```

### Anything in the logs that might be useful for us?

```txt
2023-08-15 02:54:27.493 DEBUG (MainThread) [homeassistant.components.esphome.manager] carvoltage-esp: 192.168.0.208 disconnected (expected=False), running disconnected callbacks
```

bdraco, 2023-08-15:
https://github.com/esphome/esphome/blob/b9e9223fddfa47e8641c25d4e87a7c90866606f8/esphome/components/api/api_server.cpp#L318

on_shutdown sends the shutdown and hopes it gets delivered. If the wifi connection is poor its likely never going to read HA before the connection is closed as it only does a `delay(10)` after

Jordan-Szwejda, 2023-08-19:
I increased timeout from 10 ms to 10 seconds and it looks like disconnection from HA completed successfully for my device.
```
2023-08-19 12:58:26.513 DEBUG (MainThread) [homeassistant.components.esphome.manager] carvoltage-esp: 192.168.0.208 disconnected (expected=True), running disconnected callbacks
```

Of course timeout change is just as a temp solution. The correct implementation should not relay on delay operation but rather on disconnection  completion event. If such disconnection completion event is not obtained in a specified period of time (timeout) then anyway the node should enter deep sleep.

VIMVa, 2023-08-22:
@Jordan-Szwejda
I've been looking around how to (temporarily) increase the timeout but sofar I have been unable to find the api_server.cpp file that distributes the flag towards the devices. I already found out that it's not in the 'build' folder because there it is reversed to 10 every time I  install the code.
Please note that I'm just above newbie in technical stuff.

Jordan-Szwejda, 2023-08-24:
@VIMVa I did not have time to analyse how to do it correctly - creating your own branch/pull_request and build a docker image based on this. To do it quickly I entered docker container and changed it in the filesystem of the running container. After that you can clean project from web UI and install firmware. Do not destroy running container before installing modified firmware.
Below is a video tutorial how to do it quickly on windows by using docker desktop:

[image]

I would appreciate if anyone tells me how to build a docker image based on your branch of the code.

ssieb, 2023-08-24:
You can used the modified `api` component as an external component.  No need to mess around with files in the container.

Jordan-Szwejda, 2023-09-05:
@ssieb Thanks for pointing out. I added external component to config yaml and my code  was automatically  included without messing with docker container.

VIMVa, 2023-09-05:
> @ssieb Thanks for pointing out. I added external component to config yaml and my code was automatically included without messing with docker container.

Any chance that you could describe that process so that it can be understood by a newbie in programming?

Jordan-Szwejda, 2023-09-06:
@VIMVa That's very simple. When you create a configuration yaml for a device you just add something like:

```
external_components:
  - source:
      type: local
      path: external-components
    components: [ api ]
```

path: is a relative path to your custom components. Your code should be copied into config directory

In my case directory structure is like:
```
config
  .esphome
  external-components
    api
       __pycache__
       api_connection.cpp
       api_server.cpp
       <other api files from esphome repository>
  carvoltage-esp.yaml
  esp-test.yaml
 ...
```
After that you may modify the code in components you specified in the components section of configuration. In my case it is exactly API

components: [ api ]

When you Install such firmware on a device you will have your custom changes applied.

VIMVa, 2023-09-06:
@Jordan-Szwejda
Thanks a million!!!
You just made the fog disappear.

VIMVa, 2023-09-07:
...and I managed to make it work. Increased the delay at send_disconnect_request from 10 to 100 and I don't see 'Unavailable' statuses any more for hours.
Thanks again!

elministerr, 2024-01-12:
any chance to put the increased delay in a upcoming version ? i have the same issue. Sometimes my devices going unavailable in deep sleep.. explicit a temp sensor.

scarolan, 2024-01-15:
Same problem here, have a battery powered esp32c3 that sleeps for ten minutes, wakes for fifteen seconds to send data, then goes back to deep sleep. There should be an option to allow devices to check in less frequently instead of simply marking them permanently as "unavailable".

micz, 2024-01-29:
I tried to import the api component setting the delay to 10000 as described here.
The ESP32 keeps trying to go to deep sleep and wakes up immediatly.
I fixed the problem following these instructions: https://community.home-assistant.io/t/entity-values-unavailable-during-sleep/626661/7

The first time I added the ESP32 to HomeAssistant it was without deepsleep, because I was polishing things up.
When I added the deepsleep, the sensors were always going unavailable.
Now everything works correctly.

fran6120, 2024-03-04:
First of all sorry for the off topic.

Something similar happens with the wifi.disable action that does not disconnect from home assistant gracefully.

The binary status sensor shows that it is still connected, and the rest of the entities continue to show values. After a while all entities appear unavailable.

A while ago I opened an issue: #5414

ryrun, 2025-05-16:
The problem with Deep Sleep not properly disconnecting from HA and causing the entities to become unavailable is, in my opinion, a serious issue. It’s really a shame that this has been an open topic for so long. ESPHome itself is great, but Deep Sleep seems pretty much useless. There was already this issue: https://github.com/home-assistant/core/issues/58143#issue-1031947583 — but it doesn’t really solve the root problem.

closed 2025-06-08 by jesserockz