esphome/issues #4796: Initial value of global is broken

Lumpusz, 2023-08-15:
### The problem

I would expect the following config initializes the switch as 'ON' right after the device boots, and this is indeed the case on 2023.6.5, but when I upgraded to 2023.7.1 this stopped working, and the device comes up as 'OFF'.

```
globals:
  - id: g_on
    type: bool
    initial_value: 'true'

switch:
  - platform: template
    name: "${devicename}"
    lambda: "return id(g_on);"
    turn_on_action:
      - globals.set:
          id: g_on
          value: 'true'
    turn_off_action:
      - globals.set:
          id: g_on
          value: 'false'
```

### Which version of ESPHome has the issue?

2023.7.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP8266

### Board

esp01_1m (Shelly 1PM)

### Component causing the issue

globals

### Example YAML snippet

```yaml
see above
```

### Anything in the logs that might be useful for us?

```txt
no suspicious log entry
```

cinadr, 2023-08-18:
+1 for me. How did you overcome with this?

Lumpusz, 2023-08-18:
> +1 for me. How did you overcome with this? I made some on_boot code to set my default vars to true.
> 
> ```
> esphome:
>   on_boot:
>     then:
>       - globals.set:
>           id: globalvar
>           value: "true"
> ```

I restored previous version of esphome addon frombackup.

ssieb, 2023-09-09:
The default restore mode is `always_off`, so the switch is set to off which sets the global variable.  Either change the restore mode to restore the value or disable the restore.

closed 2023-09-09 by ssieb

rmandjes, 2023-09-09:
I think that I had already enabled that:
- id: show_good_status # if false: show only on orange and red not green
    type: bool
    restore_value: true
    initial_value: "true"

But that is not working.

And according to the documentation this are the only options and there is no always_off value mentioned:
Configuration variables:

- id (Required, [ID](https://esphome.io/guides/configuration-types#config-id)): Give the global variable an ID so that you can refer to it later in [lambdas](https://esphome.io/guides/automations.html?highlight=global#config-lambda).

- type (Required, string): The C++ type of the global variable, for example bool (for true/false), int (for integers), float (for decimal numbers), int[50] for an array of 50 integers, etc.

- restore_value (Optional, boolean): Whether to try to restore the state on boot up. Be careful: on the ESP8266, you only have a total of 96 bytes available for this! Defaults to no. This will use storage in “RTC memory”, so it won’t survive a power-cycle unless you use the esp8266_restore_from_flash option to save to flash. See [esp8266_restore_from_flash](https://esphome.io/components/esphome) for details.

- initial_value (Optional, string): The value with which to initialize this variable if the state can not be restored or if state restoration is not enabled. This needs to be wrapped in quotes! Defaults to the C++ default value for this type (for example 0 for integers).

fritsjan, 2024-01-03:
same here: initial_value (for bool) is not working and setting restore_value to true or false does not do anything.
I could use the workaround setting the value on boot, but then I do not have the option to use a restore_value...
Also, as mentioned, it does not act like the documentation suggests...
Clearly a bug... please reopen this.

ssieb, 2024-01-04:
The original issue here is invalid.  If you think yours is different, then paste your config and describe what is happening.  But it would be better to come ask on the esphome discord server instead.

matthiasdg, 2024-03-31:
Why should we head over to discord? It's way easier/more logical to search for answers on GitHub. Behavior changed compared to older versions, that's a fact.
The answer was mentioned, but maybe not clear enough. It's the restore_mode of the switch, not the restore_value of the var. (https://esphome.io/components/switch/index.html; I put it on RESTORE_DEFAULT_ON now)

Salvora, 2024-04-26:
@ssieb 
I don't get this. What is the point of "initial value" and "restore value" config options in the global section if it gets overridden by something else. This was not the default behaviour pre-update for "globals" and docs don't make sense.

https://esphome.io/guides/automations.html?highlight=globals#global-variables

For me, below code was working perfectly pre-update but now it does not. 
```
globals:
  - id: display_status
    type: bool
    restore_value: true
    initial_value: "true"

  - id: display_mode_global
    type: int
    restore_value: true
    initial_value: "1"
```

ssieb, 2024-04-26:
It's restoring, but you have something else changing the value.  Probably a switch.
But this is not the place to discuss it.