esphome/issues #4800: No Graph Line

4noxx, 2023-08-16:
### The problem

I added a graph for my epaper Display, but i see only the border.... no line

### Which version of ESPHome has the issue?

2023.8.0b3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.7.3

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
sensor:
  - platform: homeassistant
    id: temp_aussen
    entity_id: sensor.foxess_ambient_temperature

graph:
  - id: temp_aussen_graph
    sensor: temp_aussen
    duration: 1h
    width: 150
    height: 50

display:
  - platform: waveshare_epaper
    id: epaperdisplay
    cs_pin: 15
    dc_pin: 12
    update_interval: 120s
    busy_pin: 16
    reset_pin: 4
    model: 4.20in
    lambda: |-
      it.graph(200,10, id(temp_aussen_graph));
```

4noxx, 2023-08-16:
[image]

ssieb, 2023-08-16:
Have you added the device to HA?  Are you sure it's getting data?  Check the logs that you were asked to include.

4noxx, 2023-08-17:
Yes, it works now. It seems that it takes some time to see the value line. 

Is it possible to "import" the values from HA? After an ESP restart the graph is empty at first.

ssieb, 2023-08-17:
There's no way to get historical data.

closed 2023-08-17 by ssieb