esphome/issues #4803: SML checksum error since merge of #5250

MrXermon, 2023-08-17:
### The problem

I initially reported a bug regarding the values of my DWS7412.2.G2 meter not being decoded correctly (#4494). Since the merge of [Fix 24 bit signed integer parsing in sml parser #5250](https://github.com/esphome/esphome/pull/5250), ESPHome shows checksum errors which causes no data to be extracted anymore.

### Which version of ESPHome has the issue?

2023.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

sml

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

sml:
  id: mysml
  uart_id: uart_bus

sensor:
  - platform: sml
    name: "SML Total In"
    sml_id: mysml
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 2
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
  - platform: sml
    name: "SML Total Out"
    sml_id: mysml
    obis_code: "1-0:2.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 2
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
  - platform: sml
    name: "SML Current Power"
    sml_id: mysml
    obis_code: "1-0:16.7.0"
    unit_of_measurement: W
[... 127 lines cut ...]
[08:48:06][W][sml:116]: Checksum error in received SML data.
[08:48:07][W][sml:116]: Checksum error in received SML data.
[08:48:08][W][sml:116]: Checksum error in received SML data.
[08:48:09][W][sml:116]: Checksum error in received SML data.
[08:48:10][W][sml:116]: Checksum error in received SML data.
[08:48:11][W][sml:116]: Checksum error in received SML data.
[08:48:12][W][sml:116]: Checksum error in received SML data.
[08:48:13][W][sml:116]: Checksum error in received SML data.
[08:48:14][W][sml:116]: Checksum error in received SML data.
[08:48:15][W][sml:116]: Checksum error in received SML data.
[08:48:16][W][sml:116]: Checksum error in received SML data.
[08:48:17][W][sml:116]: Checksum error in received SML data.
[08:48:18][W][sml:116]: Checksum error in received SML data.
[08:48:19][W][sml:116]: Checksum error in received SML data.
[08:48:20][W][sml:116]: Checksum error in received SML data.
[08:48:21][W][sml:116]: Checksum error in received SML data.
[08:48:22][W][sml:116]: Checksum error in received SML data.
[08:48:23][W][sml:116]: Checksum error in received SML data.
[08:48:24][W][sml:116]: Checksum error in received SML data.
[08:48:25][W][sml:116]: Checksum error in received SML data.
[08:48:26][W][sml:116]: Checksum error in received SML data.
[08:48:27][W][sml:116]: Checksum error in received SML data.
[08:48:28][W][sml:116]: Checksum error in received SML data.
[08:48:29][W][sml:116]: Checksum error in received SML data.
```

Lukel1412, 2023-08-17:
Same Problem here.

saschaabraham, 2023-08-17:
Same here with German Smart Reader an IR Reader. Worked nearly 8 months without any issue. Now with 202308 nothing.

Bascht74, 2023-08-17:
Is there a way to go back just for the SML component?

oliver507app, 2023-08-17:
same here after update

GLSSoftware, 2023-08-17:
Workaround found here: 
https://github.com/esphome/issues/issues/4805

alengwenus, 2023-08-17:
I think https://github.com/esphome/esphome/pull/4798 is the one to blame. Unfortunately, I really don't know how to fix this besides reverting everything. Maybe @Mat931 can have a look?

Mat931, 2023-08-17:
Found the problem. I forgot to subtract 2 from the buffer length when calculating the checksum.

Can you add this to your config and try if it works?
```yaml
external_components:
  - source: github://pr#5271
    components: [sml]
    refresh: always
```

MrXermon, 2023-08-17:
> Found the problem. I forgot to subtract 2 from the buffer length when calculating the checksum.
> 
> Can you add this to your config and try if it works?
> 
> ```yaml
> external_components:
>   - source: github://pr#5271
>     components: [sml]
>     refresh: always
> ```

Seems to be working for me, too. I do not see any warnings on my end. Logging with level DEBUG is enabled.

Skeletitor, 2023-08-17:
> 
> Seems to be working for me, too. I do not see any warnings on my end. Logging with level DEBUG is enabled.

Okay: I'll take a look at my config. 
@Mat931 Good job and tank you very much!

GLSSoftware, 2023-08-17:
Thank you very much for fixing the issue!

> 19:07:12 | [W] | [component:204] | Component sml took a long time for an operation (0.14 s).
> 19:07:12 | [W] | [component:205] | Components should block for at most 20-30ms.
> ```

I also do get these warnings. But I have them in other ESPHome components that do not use sml, too. e.g.:

[19:29:20][W][component:204]: Component ssd1306_base took a long time for an operation (0.30 s).
[19:29:20][W][component:205]: Components should block for at most 20-30ms.

I would guess that these warnings do have some other cause.

wtremmel, 2023-08-17:
Confirmed as well, works

Mat931, 2023-08-17:
The warnings might be because I changed the lookup table used in the CRC calculation to a smaller (512 bytes -> 64 bytes) but slightly slower one. This only affects the ESP8266 since the ESP32 has hardware acceleration for calculating CRCs.
If speed is really the problem we can change it back to the big lookup table.
If you see the warnings or not might also depend on how much data your meter sends.

jesserockz, 2023-08-17:
The PR fixing will be in 2023.8.1

closed 2023-08-17 by jesserockz