esphome/issues #4805: SML checksum error after update to 2023.8.0

mvdberge, 2023-08-17:
### The problem

After the update of ESPhome to 2023.8.0 I only receive the following error:

`Checksum error in received SML data`

When using manually installed ESPhome (via pip) and downgrading to the last version, everything is fine. So it must be a problem of version 2023.8.0.

### Which version of ESPHome has the issue?

2023.8.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

sml

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  tx_pin: RX
  rx_pin: TX
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1
  rx_buffer_size: 512

sml:
  id: mysml
  uart_id: uart_bus

sensor:
  - platform: sml
    name: "Gesamt Energiebezug"
    id: gesamt_energiebezug
    sml_id: mysml
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
      - throttle: 60s
```

Skeletitor, 2023-08-17:
Same Problem her.

Quick fix: use external sourcen from 'alengwenus'. Thankn you bdu

--- just add to you jaml START ---
external_components:
  - source:
      type: git
      url: https://github.com/alengwenus/esphome_components
    components: [ sml ]
--- just add to you jaml END ---

Skeletitor, 2023-08-17:
https://github.com/esphome/esphome/pull/5250

I'm pretty sure this causes the issue

Mat931, 2023-08-17:
Can you add this to your config and see if it fixes the error?

```yaml
external_components:
  - source: github://pr#5271
    components: [sml]
    refresh: always
```

jesserockz, 2023-08-17:
Duplicate of #4803

closed 2023-08-17 by jesserockz

mvdberge, 2023-08-18:
> Can you add this to your config and see if it fixes the error?
> 
> ```yaml
> external_components:
>   - source: github://pr#5271
>     components: [sml]
>     refresh: always
> ```

Yes, the workaround works. Thanks for the quick reply.

Also, thanks for the quick bugfix release of esphome 2023.8.1 which includes this fix! Really appreciate your work!

kobel2015, 2023-12-11:
It doesn't work  on me