esphome/issues #4806: PRs from employees rarely if ever fill out templates

disconn3ct, 2023-08-17:
### The problem

Many of the PRs filed by employees do not use the template. Will contributions from the community that ignore the templates be accepted as well?

Even "this is why" at the top is blank. Are these contributions actually reviewed or merely rubber-stamped?

### Which version of ESPHome has the issue?

2023.8.0

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Additional information

https://github.com/esphome/esphome/pull/5228
https://github.com/esphome/esphome/pull/5232

Community contributors seem to follow it:
https://github.com/esphome/esphome/pull/5205
https://github.com/esphome/esphome/pull/4434
https://github.com/esphome/esphome/pull/5157

kaechele, 2023-08-18:
What would a satisfactory resolution of this issue look like to you?

disconn3ct, 2023-08-20:
Enforce the same standard for all contributors. Or (rephrased, based on the current standard) reject undocumented or incomplete submissions from any source.

kaechele, 2023-08-20:
Understood.

I took a quick pass at trying to find anything documented (as you are referring to a standard) that would require anyone to even use the pull request template at all and that returned nothing so far.
In case there is in fact not a policy in place you could certainly make your case to the maintainers to implement this policy.
You might want to change the title of the issue in that case as right now I find it comes across unnecessarily accusatory towards the maintainers.

jesserockz, 2023-08-20:
I apologise. 
I have been lazy with my own PRs and I will be changing this going forward.
Thanks for bringing it up.

Jesse

closed 2023-08-20 by jesserockz

disconn3ct, 2023-08-21:
> I took a quick pass at trying to find anything documented

https://github.com/esphome/issues/blob/main/.github/ISSUE_TEMPLATE/bug_report.yml

Is the template not a standard?