esphome/issues #4809: Compile error on Esphome scripted main.cpp

coyotegd, 2023-08-18:
### The problem

A full description with code and error is located here:

https://community.home-assistant.io/t/esphome-compile-error-in-the-scripted-main-cpp/603995

### Which version of ESPHome has the issue?

2023.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP32

### Board

Lilygo Esp32-S3 T-Display AMOLED

### Component causing the issue

I believe ESPHome's "Delay()" has an issue with FastLED's.

jesserockz, 2023-08-18:
That linked post uses a whole bunch of external untested libraries. 

Closing as its not part of ESPHome codebase.

closed 2023-08-18 by jesserockz

coyotegd, 2023-08-18:
I don't mean to sound rude, but your response is as vague as a compiler error.  I thought the whole idea behind "includes" and "libraries" was to expand the code base.

 

What you are saying is, if I don't "test" the "untested libraries", this will never be fixed.  The error below suggests to me that "delay" and "delayMicroseconds" need attention.  Is this something I can do as ESPHome is open source?  Where do I start?

 

Ken Klausner

 

 

#define yield() esphome::yield()

#define millis() esphome::millis()

#define micros() esphome::micros()

#define delay(x) esphome::delay(x)                                          <-- obviously the "x" is a problem

#define delayMicroseconds(x) esphome::delayMicroseconds(x)    <--

 

 

vía desktop

 
 

Sent: Thursday, August 17, 2023 at 10:26 PM