esphome/issues #4811: Missing library cario since update ESPhome 2032.8.1 to cannot compile previously working yaml

anothermort, 2023-08-18:
### The problem

Cannot compile or update previously working yaml

### Which version of ESPHome has the issue?

ESPHome 2023.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.2

### What platform are you using?

ESP32

### Board

devkit

### Component causing the issue

cairo?

### Example YAML snippet

```yaml
esphome:
  name: octo-mini
  friendly_name: Octo_mini

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "*********************"

ota:
  password: "********************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Octo-Mini Fallback Hotspot"
    password: "************"

    

# Example configuration entry

font:
  - file: 'fonts/Roboto-Regular.ttf'
    id: font1
    size: 40
image:
  - file: mdi:lightning-bolt
    id: elec
    resize: 40x40
  - file: mdi:fire
    id: gas
    resize: 40x40  
  

time:
  - platform: sntp
    id: esptime

sensor:
  - platform: homeassistant
    id: elec_use
    entity_id: sensor.octopus_energy_electricity__current_accumulative_consumption
    internal: true
  - platform: homeassistant
    id: gas_use
    entity_id: sensor.octopus_energy_gas_current_accumulative_consumption
    internal: true

# Pins for Waveshare ePaper ESP Board
spi:
  clk_pin: GPIO13
  mosi_pin: GPIO14

# Now render everything on the ePaper screen.
display:
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: GPIO15
    dc_pin: GPIO27
    busy_pin: GPIO25
    reset_pin: GPIO26
    reset_duration: 200ms
    model: 2.90inv2
    update_interval: 1min
    # If everything works, try to set full_update_every to 30 or 60
    full_update_every: 1
    rotation: 90°
    lambda: |-
      //it.print(100, 0, id(font1), "16:27");
      // Print time in HH:MM format
      it.strftime(100, 0, id(font1),  TextAlign::BASELINE_LEFT, "%H:%M", id(esptime).now());
            // Print inside temperature (from homeassistant sensor)
      it.image(30, 45, id(elec));
      if (id(elec_use).has_state()) {
        it.printf(85, 75, id(font1), TextAlign::BASELINE_LEFT ,"%.2f kWh", id(elec_use).state);
      }        
      it.image(30, 85, id(gas));
      if (id(gas_use).has_state()) {
        it.printf(85, 120, id(font1), TextAlign::BASELINE_LEFT ,"%.2f kWh", id(gas_use).state);
      }  
captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.8.1
INFO Reading configuration /config/esphome/octo-mini.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1023, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1001, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 986, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 840, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 828, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 756, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 125, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 507, in run
    validated = schema(self.conf)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 353, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 353, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/esphome/esphome/voluptuous_schema.py", line 148, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/components/image/__init__.py", line 174, in _file_schema
    return validate_file_shorthand(value)
  File "/esphome/esphome/components/image/__init__.py", line 130, in validate_file_shorthand
    validate_cairosvg_installed(value)
  File "/esphome/esphome/components/image/__init__.py", line 80, in validate_cairosvg_installed
    import cairosvg
  File "/usr/local/lib/python3.9/dist-packages/cairosvg/__init__.py", line 26, in <module>
    from . import surface  # noqa isort:skip
  File "/usr/local/lib/python3.9/dist-packages/cairosvg/surface.py", line 9, in <module>
    import cairocffi as cairo
  File "/usr/local/lib/python3.9/dist-packages/cairocffi/__init__.py", line 47, in <module>
    cairo = dlopen(
  File "/usr/local/lib/python3.9/dist-packages/cairocffi/__init__.py", line 44, in dlopen
    raise OSError(error_message)  # pragma: no cover
OSError: no library called "cairo-2" was found
no library called "cairo" was found
no library called "libcairo-2" was found
cannot load library 'libcairo.so.2': libcairo.so.2: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libcairo.so.2'
cannot load library 'libcairo.2.dylib': libcairo.2.dylib: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libcairo.2.dylib'
cannot load library 'libcairo-2.dll': libcairo-2.dll: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libcairo-2.dll'
```

### Additional information

New to ESPhome and HA so hope this is the correct place

sblogoshs, 2023-08-18:
Have this error message also when updating but only with display ESPs.

grssll, 2023-08-18:
Same for me. Esphome 2032.8.1

```INFO ESPHome 2023.8.1
INFO Reading configuration /config/esphome/esphome-web-34f45c.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1023, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1001, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 986, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 840, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 828, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 756, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 125, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 507, in run
    validated = schema(self.conf)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 353, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 353, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/esphome/esphome/voluptuous_schema.py", line 148, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/components/image/__init__.py", line 174, in _file_schema
    return validate_file_shorthand(value)
  File "/esphome/esphome/components/image/__init__.py", line 130, in validate_file_shorthand
    validate_cairosvg_installed(value)
  File "/esphome/esphome/components/image/__init__.py", line 80, in validate_cairosvg_installed
    import cairosvg
  File "/usr/local/lib/python3.9/dist-packages/cairosvg/__init__.py", line 26, in <module>
    from . import surface  # noqa isort:skip
  File "/usr/local/lib/python3.9/dist-packages/cairosvg/surface.py", line 9, in <module>
    import cairocffi as cairo
  File "/usr/local/lib/python3.9/dist-packages/cairocffi/__init__.py", line 47, in <module>
    cairo = dlopen(
  File "/usr/local/lib/python3.9/dist-packages/cairocffi/__init__.py", line 44, in dlopen
    raise OSError(error_message)  # pragma: no cover
OSError: no library called "cairo-2" was found
no library called "cairo" was found
no library called "libcairo-2" was found
cannot load library 'libcairo.so.2': libcairo.so.2: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libcairo.so.2'
cannot load library 'libcairo.2.dylib': libcairo.2.dylib: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libcairo.2.dylib'
cannot load library 'libcairo-2.dll': libcairo-2.dll: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'libcairo-2.dll'```
`

grssll, 2023-08-18:
> Have this error message also when updating but only with display ESPs.

You are right. I've removed the display and updated successfully

anothermort, 2023-08-18:
> Have this error message also when updating but only with display ESPs.

Yes seems to be display / font related .
Removed display & font from yaml and could upgrade

Same issue after editting yaml and putting fonts and display back in.

grssll, 2023-08-18:
Solution, remove gfonts and do local font
```font:
  - file: "Roboto-Bold.ttf"
    id: bold
    size: 60```

sblogoshs, 2023-08-18:
Would be nice, except I don't use gfonds, only locale.

grssll, 2023-08-18:
> Would be nice, except I don't use gfonds, only locale.

Worked for me. Try to change to different fonts maybe?

sblogoshs, 2023-08-18:
Have gone through all my local fonts and none work, must be another solution, worked before.

CountParadox, 2023-08-18:
I get the same error after changing to local fonts

anothermort, 2023-08-18:
Yes same here use local fonts. Tried changing to a default local icon as well.

sandzilla, 2023-08-18:
Same issue here. Tried switching fonts, didn't work.

Another thing (probably not related) is that on 2023.7.x there was a bug (with the animation component) that required insertion of some additional yaml to get it to compile successfully. I tried removing this code, but the issue still persists.

okoehler, 2023-08-18:
I get the same issue when trying to use Material Design icons

```
image:
  - file: mdi:alert-outline
    id: alert
    resize: 80x80
```

HeroCity99, 2023-08-18:
I am having the same issue. after playing around with my yaml file, I found out the mdi files are causing the issue. 
If you remove all your mdi files then it should be ok. The display or fonts don't cause any problems (at least for me).
reported it on Discord and I got this reply back from the developers:

"Until the container is fixed, you would have to get into the container and install the package yourself.
Apt update
Apt install ..."

sblogoshs, 2023-08-18:
To remove all mdi-files can not be the solution, was only introduced with one of the previous update, so important are the innovations in 2023.8.1 not that I can not wait until the bug is fixed. With the other ESPs there are no problems.

fusionstream, 2023-08-19:
Does anyone know how to get into the container?

So far I've gone through
1. I had to disable **Protection Mode** in the addon **Advanced SSH & Web Terminal** and then restart it
2. Then I SSHed in as per normal.
3. Then I had to enter this command:
`ssh -t hassio@localhost -p 22 "docker exec -it $(docker ps -f name=homeassistant -q) bash"`
in order to gain access to the actual os
I'm using port 22 and with the SSH addon as I understand it the root user becomes hassio

But then I'm just returned to the main shell (albeit a new instance) where apt does not exist. 

I tried `apk add apt` but app is still not in my path

I installed all the cairo packages from `apk list | grep cairo` and restarted everything and it still wont compile.

Any ideas?

ssieb, 2023-08-19:
Fixed by https://github.com/esphome/esphome/pull/5276.  It will be in the next point release shortly.

closed 2023-08-19 by ssieb