esphome/issues #4812: LD2410

percy10442, 2023-08-18:
### The problem

If I write the number in yaml about LD2410,  it can't connect HA API.

number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold

### Which version of ESPHome has the issue?

2023.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8

### What platform are you using?

ESP8266

### Board

D1 mini

### Example YAML snippet

```yaml
esphome:
  name: master-bedroom
esp8266:
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0 # disable serial logging; pins are used for sensor module comm

# Enable Home Assistant API
api:
  encryption:
    key: "G3xZP536csBt/gNjOvdaevQUYNBAChoXP516g2MlUan="

ota:
  password: "6490309c2967b254ba2cc4580d284e5e"

wifi:
  ssid: "PChome"
  password: "1111111111"
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "master-bedroom"
    password: "1111111111111"
    
web_server:
  port: 80
  
captive_portal:

i2c:
  sda: D2
  scl: D1
  scan: true
  id: bus_a

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:

sensor:
  - platform: bh1750
    name: "master bedroom illuminance"
    id: ill
    address: 0x23
    update_interval: 1s
  - platform: sht3xd
    temperature:
      name: "master bedroom temperature"
      id: temperature
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 35.8 -> 28.08
            - 36.5 -> 28.78
    humidity:
      name: "master bedroom humidity"
      id: humidity
      filters:
        - calibrate_linear:
            # Map 0.0 (from sensor) to 0.0 (true value)
            - 50.3 -> 77.87
            - 53.6 -> 81.17
    address: 0x44
    update_interval: 1s
  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
    g0:
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1:
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2:
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3:
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4:
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5:
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6:
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7:
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8:
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy

number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold

switch:
  - platform: ld2410
    engineering_mode:
      name: "engineering mode"

text_sensor:
  - platform: ld2410
    version:
      name: "firmware version"
    mac_address:
      name: "mac address"
      
binary_sensor:
  - platform: gpio
    pin: D6
    name: "master bedroom motion"
    id: motion
    device_class: motion
  - platform: ld2410
    has_target:
      name: master bedroom presence
    has_moving_target:
      name: master bedroom Moving Target
    has_still_target:
      name: master bedroom Still Target  

mqtt:
  broker: homeassistant.local
  username: pchome
  password: 11111111

interval:
  - interval: 1min
    then:
      - mqtt.publish:
          topic: IRminic328/ble/sensor/
          payload: !lambda |-
              return to_string(id(temperature).state);
```

ssieb, 2023-08-18:
It will be very difficult to determine the problem without serial logs.

percy10442, 2023-08-18:
> It will be very difficult to determine the problem without serial logs.

1. The webserver show the below information.
15:50:34 | [E] | [json:039] | Could not allocate memory for JSON document! Requested 408 bytes, largest free heap block: 408 bytes
-- | -- | -- | --
15:50:34 | [E] | [json:039] | Could not allocate memory for JSON document! Requested 240 bytes, largest free heap block: 240 bytes
15:50:34 | [E] | [json:039] | Could not allocate memory for JSON document! Requested 240 bytes, largest free heap block: 240 bytes
15:50:34 | [E] | [json:039] | Could not allocate memory for JSON document! Requested 192 bytes, largest free heap block: 192 bytes

2. The logs show the below information.
INFO ESPHome 2023.8.1
INFO Reading configuration /config/esphome/master-bedroom.yaml...
INFO Starting log output from master-bedroom.local using esphome API
WARNING Can't connect to ESPHome API for master-bedroom.local: Error connecting to ('192.168.0.99', 6053): [Errno 111] Connect call failed ('192.168.0.99', 6053) (SocketAPIError)
INFO Trying to connect to master-bedroom.local in the background

ssieb, 2023-08-18:
Oh, I didn't notice that you have the web server as well.  You might have too many things for an 8266.  Do you need the web server?  Try removing it.

percy10442, 2023-08-18:
> Oh, I didn't notice that you have the web server as well. You might have too many things for an 8266. Do you need the web server? Try removing it.

I had tried removing it, but it has the same issue.

percy10442, 2023-09-13:
As a result, I removed MQTT publishing and ESPhome ran normally.