esphome/issues #4814: Crash after connecting to MQTT

Mostalk, 2023-08-19:
### The problem

Reboots began after I started working with a binary sensor to turn off sleep mode

I had the code below and everything was fine, but after I added `publish_initial_state: True`, restarts began, I tried to completely remove all the code associated with disabling sleep mode, but without success

Of course, I'm not 100% sure that reboots are related to this, but this is the only configuration change that I can link errors to

Old part code with binary_sensor:

```
binary_sensor:
  - platform: homeassistant
    id: prevent_deep_sleep
    **// publish_initial_state: True <- after add this line reboots started** 
    entity_id: input_boolean.deep_sleep
    on_press:
       then:
          - deep_sleep.prevent: sleep_id

script:
  - id: consider_deep_sleep
    mode: queued
    then:
      - delay: 15s
      - if:
          condition:
            binary_sensor.is_on: prevent_deep_sleep
          then:
            - deep_sleep.prevent: sleep_id
            - logger.log: 'Skipping sleep, per prevent_deep_sleep'
          else:
            - deep_sleep.enter: sleep_id 
      - script.execute: consider_deep_sleep
```

### Which version of ESPHome has the issue?

2023.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP32

### Board

ESP32S V1.1

### Component causing the issue

MQTT

### Example YAML snippet

```yaml
esphome:
  name: esp32
  friendly_name: ESP32

esp32:
  board: esp32dev
    
i2c:
  - id: bus_a
    sda: 16
    scl: 17
    scan: true
  
  - id: bus_b
    frequency: 200kHz
    sda: 18
    scl: 19
    scan: true

deep_sleep:
  id: sleep_id
  run_duration: 30s
  sleep_duration: 5min

logger:

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

/* Some sensors */ 

mqtt:
  broker: 192.168.15.195
  username: homeassistant
  password: !secret mqtt_pass
  birth_message:
  will_message:

time:
  - platform: sntp
    id: my_time

// The binary sensor was here
```

### Anything in the logs that might be useful for us?

```txt
[13:56:31][I][wifi:557]: WiFi Connected!
[13:56:31][C][api:025]: Setting up Home Assistant API server...
[13:56:31][C][mqtt:037]: Setting up MQTT...
[13:56:31][I][mqtt:246]: Connecting to MQTT...
[13:56:31][I][mqtt:286]: MQTT Connected!
[13:56:36]E (28105) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[13:56:36]E (28105) task_wdt:  - loopTask (CPU 1)
[13:56:36]E (28105) task_wdt: Tasks currently running:
[13:56:36]E (28105) task_wdt: CPU 0: IDLE
[13:56:36]E (28105) task_wdt: CPU 1: IDLE
[13:56:36]E (28105) task_wdt: Aborting.
[13:56:36]
[13:56:36]abort() was called at PC 0x400f9bb8 on core 0
[13:56:36]
[13:56:36]
[13:56:36]Backtrace:0x40083d71:0x3ffbeaac |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40083d71: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
[13:56:36]
[13:56:36]
[13:56:36]
[13:56:36]
[13:56:36]ELF file SHA256: 0000000000000000
[13:56:36]
[13:56:36]Rebooting...
[13:56:36]ets Jul 29 2019 12:21:46
[13:56:36]
[13:56:36]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:56:37]configsip: 0, SPIWP:0xee
[13:56:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:56:37]mode:DIO, clock div:2
[13:56:37]load:0x3fff0030,len:1184
[13:56:37]load:0x40078000,len:13132
[13:56:37]load:0x40080400,len:3036
[13:56:37]entry 0x400805e4
```

Mostalk, 2023-08-19:
If I turn off MQTT in the configuration, then everything works as it should

akomelj, 2023-08-20:
I am having the exact same issue with 2023.8.1 version on both ESP32 Arduino and ESP-IDF platforms. I wanted to investigate the issue and strangely enough - when changing the global log level to DEBUG - crashes completely disappeared. 🤷

closed 2024-01-03 by github-actions[bot]