esphome/issues #4816: HOSTNAME does not display in DHCP leases. network scans etc.

deanfourie1, 2023-08-19:
### The problem

The hostname of my nodes just will not display on any DHCP lease information or any network scans.

It just displays the vendor, "espressif inc".

Is there a way to display the device "name" or "friendly-name" as the hostname. 

When I open the logs, it displays the correct hostname, however this does not display.

It does however work find on a ESP01_1m board but does not seem to work on any of my ESP32dev boards.

### Which version of ESPHome has the issue?

2023.8.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

networking/mdns

### Example YAML snippet

```yaml
esphome:
  name: pool-outputs
  friendly_name: "pool-outputs"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

# Example configuration entry for ESP32
i2c:
  sda: 4
  scl: 5
  scan: true
  id: bus_a

# Example configuration entry
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0

  # Optional manual IP
  manual_ip:
    static_ip: *********
    gateway: *********
    subnet: *********

# Example configuration entry
pcf8574:
  - id: 'pcf8574_hub_out_1'  # for output channel 1-8
    address: 0x24

  - id: 'pcf8574_hub_in_1'  # for input channel 1-8
    address: 0x22

# Individual outputs
switch:
  - platform: gpio
    name: "a8-light1"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 0
      mode: OUTPUT
      inverted: true

  - platform: gpio
    name: "a8-light2"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 1
      mode: OUTPUT
      inverted: true
     
  - platform: gpio
    name: "a8-light3"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 2
      mode: OUTPUT
      inverted: true
     
  - platform: gpio
    name: "PumpAlarm"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 3
      mode: OUTPUT
      inverted: true
     
  - platform: gpio
    name: "TopStairLights"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 4
      mode: OUTPUT
      inverted: true
     
  - platform: gpio
    name: "BottomWallLights"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 5
      mode: OUTPUT
      inverted: true

  - platform: gpio
    name: "TopWallLights"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 6
      mode: OUTPUT
      inverted: true
     
  - platform: gpio
    name: "PoolPump"
    pin:
      pcf8574: pcf8574_hub_out_1
      number: 7
      mode: OUTPUT
      inverted: true
     
binary_sensor:
  - platform: gpio
    name: "a8-input1"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 0
      mode: INPUT
      inverted: true

  - platform: gpio
    name: "a8-input2"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 1
      mode: INPUT
      inverted: true

  - platform: gpio
    name: "a8-input3"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 2
      mode: INPUT
      inverted: true

  - platform: gpio
    name: "a8-input4"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 3
      mode: INPUT
      inverted: true

  - platform: gpio
    name: "a8-input5"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 4
      mode: INPUT
      inverted: true

  - platform: gpio
    device_class: door
    name: "PoolCabinetTamper"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 5
      mode: INPUT

  - platform: gpio
    device_class: door
    name: "PoolCover"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 6
      mode: INPUT

  - platform: gpio
    name: "PumpStatus"
    pin:
      pcf8574: pcf8574_hub_in_1
      number: 7
      mode: INPUT
      inverted: true

  - platform: gpio
    name: "a8-s3"
    pin:
      number: 32
      inverted: true

  - platform: gpio
    name: "a8-s4"
    pin:
      number: 33
      inverted: true
     
dallas:
  - pin: 14

sensor:
  - platform: dallas
    address: 0x0c0121128d004c28
    name: "PoolTemperature"

  - platform: aht10
    temperature:
      name: "PoolCabinetTemperature"
    humidity:
      name: "PoolCabinetHumidity"
    update_interval: 60s
```

randybb, 2023-08-19:
On my Unifi controller I see the same hostnames as I have in my config under
```yaml
esphome:
  name: blabla
```
`friendly_name:` is only an internal name that gets prefixed to all entities.

ssieb, 2023-08-20:
It works for wifi, but maybe there's an issue with how dhcp requests work with ethernet that the hostname is missing.

randybb, 2023-08-20:
Have only one device with ethernet and its hostname is visible (just doblechecked that). But in the config you have a manual configuration so DHCP has nothing to do with this device.

ssieb, 2023-08-20:
Oh, good point!

closed 2023-08-20 by ssieb

deanfourie1, 2023-08-20:
Yea nah I have both name and friendly name in my configs.

I have 1 or 2 devices displaying the hostname, the rest don't.

I've gone through the configs but cannot find any thing different between the configs.

ssieb, 2023-08-20:
It's up to the network equipment to determine the name and I don't know how it would do it without a dhcp request.  But it's not an esphome issue.