esphome/issues #4819: LibreTiny ESPHome on Home Assistant won't start

droidsteve, 2023-08-21:
### The problem

installed add-on
click start, and "start" just spins

### Which version of ESPHome has the issue?

LibreTiny ESPHome Current version: 2023.9.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
Log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: LibreTiny ESPHome
 LibreTiny ESPHome add-on for intelligently managing all your ESP8266/ESP32/LibreTiny devices
-----------------------------------------------------------
 Add-on version: 2023.9.0-dev
 You are running the latest version of this add-on.
 System: Home Assistant OS 10.4  (amd64 / qemux86-64)
 Home Assistant Core: 2023.8.3
 Home Assistant Supervisor: 2023.08.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[19:11:21] INFO: Configuring timezone (America/New_York)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-nginx: starting
s6-rc: info: service esphome: starting
s6-rc: info: service esphome successfully started
s6-rc: info: service init-nginx successfully started
s6-rc: info: service nginx: starting
s6-rc: info: service nginx successfully started
s6-rc: info: service discovery: starting
[19:11:21] INFO: Waiting for ESPHome dashboard to come up...
[19:11:21] INFO: Starting ESPHome dashboard...
2023-08-20 19:11:21,963 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/libretuya-esphome...
[19:11:22] INFO: Starting NGINX...
[19:11:22] ERROR: Got unexpected response from the API: Add-ons must list services they provide via discovery in their config!
[19:11:22] INFO: Successfully send discovery information to Home Assistant.
s6-rc: info: service discovery successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
```

jesserockz, 2023-08-21:
Please open libretiny issues on the libretiny repos until it is properly merged and supported

closed 2023-08-21 by jesserockz

droidsteve, 2023-08-21:
their github sent me here, will look again...