esphome/issues #4820: compile issue for ESP32s on ARM Mac

thatslolo, 2023-08-21:
### The problem

Since updating to 2023.08 I get the following error when trying to compile on a manual (native pip) install.

I force reinstalled esphome already and removed platformio to get a fresh install also. No luck for now.

### Which version of ESPHome has the issue?

2023.08.2

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

----------------------

esphome:
  friendly_name: ${friendly_name}
  name: ${slug}
  # 160 Mhz Board
  platformio_options:
    board_build.f_cpu: 160000000L

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      # 160 Mhz Board
      CONFIG_ESP32_DEFAULT_CPU_FREQ_160: y
      CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ: "160"
```

### Anything in the logs that might be useful for us?

```txt
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Building ESP-IDF components for target esp32c3
-- Configuring incomplete, errors occurred!
See also "…/Git-Projects/ESPHome-YAML/.esphome/build/deckenlampe-schlafzimmer/.pioenvs/deckenlampe-schlafzimmer/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at …/.platformio/packages/framework-espidf/tools/cmake/build.cmake:463 (message):
  Traceback (most recent call last):

    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/requests/compat.py", line 11, in <module>
      import chardet

  ModuleNotFoundError: No module named 'chardet'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/idf_component_manager/prepare_components/__main__.py", line 4, in <module>
      from .prepare import main
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/idf_component_manager/prepare_components/prepare.py", line 18, in <module>
      from ..core import ComponentManager
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/idf_component_manager/core.py", line 17, in <module>
      import requests
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/requests/__init__.py", line 45, in <module>
      from .exceptions import RequestsDependencyWarning
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/requests/exceptions.py", line 9, in <module>
      from .compat import JSONDecodeError as CompatJSONDecodeError
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/requests/compat.py", line 13, in <module>
      import charset_normalizer as chardet
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/charset_normalizer/__init__.py", line 24, in <module>
      from .api import from_bytes, from_fp, from_path, is_binary
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/charset_normalizer/api.py", line 5, in <module>
      from .cd import (
    File "…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/charset_normalizer/cd.py", line 9, in <module>
      from .md import is_suspiciously_successive_range

  ImportError:
  dlopen(…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/charset_normalizer/md.cpython-311-darwin.so,
  0x0002): tried:
  '…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/charset_normalizer/md.cpython-311-darwin.so'
  (mach-o file, but is an incompatible architecture (have 'arm64', need
  'x86_64')),
  '/System/Volumes/Preboot/Cryptexes/OS…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/charset_normalizer/md.cpython-311-darwin.so'
  (no such file),
  '…/.platformio/penv/.espidf-4.4.5/lib/python3.11/site-packages/charset_normalizer/md.cpython-311-darwin.so'
  (mach-o file, but is an incompatible architecture (have 'arm64', need
  'x86_64'))

Call Stack (most recent call first):
  …/.platformio/packages/framework-espidf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.27 seconds ==========================
```

ModuCZ, 2023-08-24:
Solved this issue by cleaning up ESPHome work dir. For smaller environment, using "Clean Build Files" under 3 dots at Dashboard might help.

I'm lazy and my environment is quite big so i solved it this way:
1. Stop esphome container
2. Make backup of whole "esphome" directory
3. Delete whole ".esphome" folder (dont touch .yaml files)
4. Start esphome container
5. Use "Update all" function
6. Problem solved

I'm using ESPHome in Docker environment, this might not work for other solutions.

thatslolo, 2023-08-24:
Solved the issue by installing `chardet` manually in the pio penv for espidf.

`~/.platformio/penv/.espidf-4.4.5/bin/python -m pip install chardet
`

For reference:
https://community.platformio.org/t/build-failed-with-latest-pio/32013/6

closed 2023-08-24 by thatslolo