esphome/issues #4822: Can't use IO12 and IO13 status leds for LuatOS/Core ESP32C3 boards

samuelolteanu, 2023-08-22:
### The problem

I've got two of these ESP32C3 boards from Aliexpress. Small issue with them: according to their [documentation](https://wiki.luatos.com/chips/esp32c3/board.html), the two red leds that work fine with their shipped firmware, (they flash alternatively on power up) are on (GP)IO12 and (GP)IO13 and cannot be used because they are reserved by SPI/PSRAM interface. Any possible override so i can test them out?
Flashed with esphome or tasmota, those leds will permanently stay barely lit.
[image]
[image]

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Board

ESP32C3 CORE

### Component causing the issue

status_led

### Additional information

They won't boot with esphome default config. Using arduino framework, usb serial will hang out when trying to read when not in bootloader mode. I needed to add "board_build.flash_mode: dio" to platformio_options even though it's got 4mb of flash.

jesserockz, 2023-08-22:
If the board pin map is added to https://github.com/esphome/esphome/blob/dev/esphome/components/esp32/boards.py#L44 for this specific `board` then that overrides the pin blocking in ESPHome.

zwgmy, 2023-09-02:
following:
https://github.com/esphome/issues/issues/4454

tailorcai, 2023-11-24:
my workaround was using a shadow template output and lambda:
```
  on_boot:
    priority: 600
    then:
      - lambda: |-
          gpio_set_direction(GPIO_NUM_12, GPIO_MODE_OUTPUT);
          gpio_set_direction(GPIO_NUM_13, GPIO_MODE_OUTPUT);

output:
  - platform: template
    id: blink_led
    type: binary
    write_action:
      - lambda: |-
          gpio_set_level(GPIO_NUM_12, state);

```

closed 2023-12-21 by jesserockz