esphome/issues #4828: Pipsolar: Checksum calculation broken - causes POP02 (set output priority to SBU) to fail with NAK

philpem, 2023-08-23:
### The problem

After updating ESPHome, my solar inverter automation (which uses Pipsolar) stopped working.

On further investigation, it sent a POP02 command to the inverter (process -> output priority -> 02/SBU) but the outgoing command had an invalid checksum.
The POP00 and POP01 commands are sent with a correct checksum:

```
[12:48:32][D][uart_debug:114]: >>> 50:4F:50:30:32:E2:0A:0D  (note: this is the POP02 command)
[12:48:32][D][pipsolar:808]: Sending command from queue: POP02 with length 5
[12:48:32][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D     (note: this is a "NAK", negative ack, bad CRC)
[12:48:32][D][pipsolar:043]: response length for command OK
[12:48:32][D][pipsolar:773]: checking crc on incoming message
[12:48:32][D][pipsolar:776]: CRC OK
[12:48:32][D][pipsolar:049]: command not successful
```

Digging deeper I found this changeset which went in a few weeks ago:

https://github.com/esphome/esphome/commit/98bf4276000e39f22f4b2977fe7ba4d05609a25c

This removes the Pipsolar-specific CRC function calls (and the functions themselves) and replaces them with a standard "crc16be" CRC function.

The problem is the Pipsolar (MPPSolar, Voltronic, Easun, etc) inverters don't allow the bytes 0x0D, 0x0A or 0x28 in the CRC, and convert them to 0x0E, 0x0B or 0x29 respectively before transmitting (or checking) the CRC. This is normally handled by this block of code, which is part of Pipsolar.

```
  if (b_crc_low == 0x28 || b_crc_low == 0x0d || b_crc_low == 0x0a)
    b_crc_low++;
  if (b_crc_hign == 0x28 || b_crc_hign == 0x0d || b_crc_hign == 0x0a)
    b_crc_hign++;
```

The removal of this block has broken sending of some commands in Pipsolar -- specifically `POP00` works, but `POP02` fails. 

Some incoming messages will also incorrectly be marked as having a bad CRC -- I think there's a separate bug in Pipsolar as I'm also seeing the "apparent power" being set to the same value as the "real power", which should not be the case as the inverter load is not purely resistive.

Tagging @mat931 and @jesserockz as they're the authors of the above commit.

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP8266

### Board

esp32dev

### Component causing the issue

pipsolar

### Anything in the logs that might be useful for us?

```txt
[12:48:32][D][uart_debug:114]: >>> 50:4F:50:30:32:E2:0A:0D  (note: this is the POP02 command)
[12:48:32][D][pipsolar:808]: Sending command from queue: POP02 with length 5
[12:48:32][D][uart_debug:114]: <<< 28:4E:41:4B:73:73:0D     (note: this is a "NAK", negative ack, bad CRC)
[12:48:32][D][pipsolar:043]: response length for command OK
[12:48:32][D][pipsolar:773]: checking crc on incoming message
[12:48:32][D][pipsolar:776]: CRC OK
[12:48:32][D][pipsolar:049]: command not successful
```

Note that the correct CRC for the `POP02` command is `E2 0B` due to the disallowing of bytes `0A`, `0D` and `28` in CRCs.
```

ninlee, 2023-08-27:
external_components:
  - source: github://pr#5299
    components: [pipsolar]
    refresh: always

this seemed to have fixed my problem... thanks

closed 2023-09-02 by jesserockz

ligius-, 2023-11-23:
Maybe I'm doing something wrong, but, with this branch, I am able to receive some sensors but not the others:

For example, grid_rating_current works but not grid_voltage:
```
[02:12:51][D][pipsolar:756]: timeout command to poll: QPIGS
[02:12:51][D][pipsolar:837]: Sending polling command : QPIRI with length 5
[02:12:52][D][pipsolar:773]: checking crc on incoming message
[02:12:52][D][pipsolar:776]: CRC OK
[02:12:52][D][pipsolar:430]: Decode QPIRI
[02:12:52][V][sensor:043]: 'inverter0_grid_rating_current': Received new state 26.000000
[02:12:52][D][sensor:093]: 'inverter0_grid_rating_current': Sending state 26.00000 A with 1 decimals of accuracy
[02:12:52][D][pipsolar:837]: Sending polling command : QPIGS with length 5
[02:12:53][D][pipsolar:773]: checking crc on incoming message
[02:12:53][D][pipsolar:782]: CRC NOK expected: D6 64 but got: 17 96
```