esphome/issues #4831: ESP32C3 no IPv6 using Arduino Framework

wtremmel, 2023-08-24:
### The problem

ESP32 does not get an IPv6 address assigned, either log output is missing or address is really not assigned

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP32

### Board

Wemos C3 Mini

### Component causing the issue

network

### Example YAML snippet

```yaml
esphome:
  name: arbeitszimmer-sensor
  platformio_options: 
    board_build.extra_flags:
      - "-DARDUINO_USB_CDC_ON_BOOT=0"

esp32:
  board: lolin_c3_mini
  framework:
    type: arduino

network:
  enable_ipv6: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 8.5

logger:
  level: VERY_VERBOSE

debug: 
  update_interval: 120s
```

### Anything in the logs that might be useful for us?

```txt
I will upload the complete log file
```

wtremmel, 2023-08-24:
[logs_arbeitszimmer-sensor_run.txt](https://github.com/esphome/issues/files/12427594/logs_arbeitszimmer-sensor_run.txt)

optimusprimespace, 2023-08-26:
What router settings are you using with IPv6? Also is IPv6 working with other devices on the same subnet? Try pinging the arbeitszimmer-sensor.local mDNS to see if it's pinging the IPv4 or IPv6 address. I recommend using SLAAC and stateless DHCP with IPv6 on the router level. My problem is that i cannot turn it off :) using the `enable_ipv6: false` command.

wtremmel, 2023-08-26:
ok, I used tcpdump on my firewall to check for communcation between the firewall and the device:
```
15:52:41.308437 IP6 fe80::6ab6:b3ff:fe75:8cb8.mdns > ff02::fb.mdns: 0 [3q] [3n] ANY (QM)? arbeitszimmer-sensor._esphomelib._tcp.local. ANY (QM)? arbeitszimmer-sensor.local. ANY (QM)? arbeitszimmer-sensor.local. (221)
15:52:41.609418 IP6 fe80::6ab6:b3ff:fe75:8cb8.mdns > ff02::fb.mdns: 0 [3q] [3n] ANY (QM)? arbeitszimmer-sensor._esphomelib._tcp.local. ANY (QM)? arbeitszimmer-sensor.local. ANY (QM)? arbeitszimmer-sensor.local. (221)
15:52:41.909345 IP6 fe80::6ab6:b3ff:fe75:8cb8.mdns > ff02::fb.mdns: 0*- [0q] 6/0/0 PTR _esphomelib._tcp.local., PTR arbeitszimmer-sensor._esphomelib._tcp.local., (Cache flush) SRV arbeitszimmer-sensor.local.:6053 0 0, (Cache flush) TXT "api_encryption=Noise_NNpsk0_25519_ChaChaPoly_SHA256" "network=wifi" "board=lolin_c3_mini" "platform=ESP32" "mac=68b6b3758cb8" "version=2023.8.2", (Cache flush) A 192.168.24.217, (Cache flush) AAAA fe80::6ab6:b3ff:fe75:8cb8 (384)
```

So with that I get the link local address `fe80::6ab6:b3ff:fe75:8cb8` which I also cain ping:
```
tremmel@lancre:~$ ping6 fe80::6ab6:b3ff:fe75:8cb8%eth0.24
PING fe80::6ab6:b3ff:fe75:8cb8%eth0.24(fe80::6ab6:b3ff:fe75:8cb8%eth0.24) 56 data bytes
64 bytes from fe80::6ab6:b3ff:fe75:8cb8%eth0.24: icmp_seq=1 ttl=255 time=58.3 ms
64 bytes from fe80::6ab6:b3ff:fe75:8cb8%eth0.24: icmp_seq=2 ttl=255 time=78.6 ms
64 bytes from fe80::6ab6:b3ff:fe75:8cb8%eth0.24: icmp_seq=3 ttl=255 time=103 ms
64 bytes from fe80::6ab6:b3ff:fe75:8cb8%eth0.24: icmp_seq=4 ttl=255 time=21.8 ms
```

Trying to guess the ULA for the sensor, this should be `fdd1:9494:7978:18:6ab6:b3ff:fe75:8cb8`:
ping also works

The global IPv6 address should be `2003:de:272f:bdc4:6ab6:b3ff:fe75:8cb8`:
also pings.

So it *does* get an IPv6 address, it is just nowhere shown in the DEBUG output. So I perhaps should rename the issue.

optimusprimespace, 2023-08-26:
You can close the issue, and also delete the global IPv6 address, because that is accessible from the outside and somebody can look up where are you located.

wtremmel, 2023-08-26:
Issue closed. Dynamic domestic IPv6, so will rotate soon anyway.

closed 2023-08-26 by wtremmel