esphome/issues #4832: ld2410 component switch not compiling

elminos, 2023-08-24:
### The problem

Following [this issue](https://github.com/esphome/esphome/pull/4434#issue-1583315048) now we can edit the settings of the sensor online while having engineering mode enabled.

As explained in the [documentation](https://esphome.io/components/sensor/ld2410.html#switch) we have to enable the engineering mode switch. As soon as I put:
```
switch:
  - platform: ld2410
    engineering_mode:
      name: "engineering mode"
    bluetooth:
      name: "control bluetooth"
```
The .yaml throws these errors in the log while compiling:
```
undefined reference to `esphome::switch_::Switch::turn_on()'
undefined reference to `esphome::switch_::Switch::turn_off()'
```

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ld2410

### Example YAML snippet

```yaml
esphome:
  name: presencia-comedor
  friendly_name: Presencia-comedor

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
#  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: REPLACEME

ota:
  password: REPLACEME

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
[... 249 lines cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/esphome/components/ld2410/ld2410.o: in function `esphome::ld2410::LD2410Component::dump_config()':
/config/esphome/.esphome/build/presencia-comedor/src/esphome/components/ld2410/ld2410.cpp:27: undefined reference to `esphome::switch_::log_switch(char const*, char const*, char const*, esphome::switch_::Switch*)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/esphome/components/ld2410/ld2410.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()':
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:657: undefined reference to `esphome::switch_::log_switch(char const*, char const*, char const*, esphome::switch_::Switch*)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/esphome/components/ld2410/ld2410.o: in function `esphome::ld2410::LD2410Component::handle_ack_data_(unsigned char*, int)':
/config/esphome/.esphome/build/presencia-comedor/src/esphome/components/ld2410/ld2410.cpp:410: undefined reference to `esphome::switch_::Switch::publish_state(bool)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/main.o:(.literal._Z5setupv+0x98): undefined reference to `vtable for esphome::ld2410::EngineeringModeSwitch'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/main.o:(.literal._Z5setupv+0xb4): undefined reference to `vtable for esphome::ld2410::BluetoothSwitch'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/main.o:(.literal._Z5setupv+0x390): undefined reference to `esphome::switch_::Switch::Switch()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/main.o: in function `setup()':
/config/esphome/.esphome/build/presencia-comedor/src/main.cpp:233: undefined reference to `esphome::switch_::Switch::Switch()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/presencia-comedor/src/main.cpp:257: undefined reference to `esphome::switch_::Switch::Switch()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/esphome/core/controller.o:(.literal._ZN7esphome10Controller16setup_controllerEb+0x20): undefined reference to `esphome::switch_::Switch::add_on_state_callback(std::function<void (bool)>&&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/presencia-comedor/.pioenvs/presencia-comedor/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/config/esphome/.esphome/build/presencia-comedor/src/esphome/core/controller.cpp:35: undefined reference to `esphome::switch_::Switch::add_on_state_callback(std::function<void (bool)>&&)'
collect2: error: ld returned 1 exit status
*** [/data/presencia-comedor/.pioenvs/presencia-comedor/firmware.elf] Error 1
========================= [FAILED] Took 164.79 seconds =========================
```

### Additional information

I'm sorry if I'm missing something or this is not the right way to report the problem. 
This is my first bug report.
Thank you for all the hard work.

sebcaps, 2023-08-26:
Hello,
  With 
``` yaml
esp32:
  board: esp32dev
  framework:
    type: arduino
```
compilation is successful.
Not skilled enough to get all implications switching from `esp-idf` to `arduino` but you may give a try. 
In the same time, not sure it's related to LD2410 component or switch component.

My 2cent's

SodaWithoutSparkles, 2023-08-29:
I found updating the docker container of esphome solved my issue of the ld2410 components not found. I want to document it here cuz this this the first google search github result.

Usually when stuff does not compile I just wipe the cache and starts again. works 90% of the time. I also uses the `espidf` framework.

also, iirc, switching frameworks cannot be done via ota. you need to physically connect it.

elminos, 2023-08-29:
Thank you guys for the feedback.
I'm really a rookie here and I'm using Home Assistant OS with the native ESPHome integration.

Following your wipe cache approach I tried the "Clean Build Files" option and now it compiles:
[image]

Not sure if this will happen to more people nor if it's an issue to be addressed or not...

All the best!

SodaWithoutSparkles, 2023-08-29:
@elminos if you have solved your issue, mark it as close please.