esphome/issues #4833: OTA Update does not work from web_server

jmr61, 2023-08-25:
### The problem

OTA Update begins, count until 85% at least then stop with message "Update succesfull", ESP reboot but update is not effective

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.4

### What platform are you using?

ESP32

### Board

Espressif ESP32-S3-DevKitC-1-N8

### Component causing the issue

OTA

### Example YAML snippet

```yaml
esphome:
  name: "test-ota"

esp32:
  board: esp32-s3-devkitc-1
  framework: 
    type: arduino
  
button:
  - platform: template
    name: "OTA 2"
   
# Enable logging
logger:  
  level: VERBOSE

ota:

api:

web_server:
  port: 80

wifi:
  ssid: test
  password: my_password

  manual_ip:
    static_ip: 192.168.1.100
    gateway: 192.168.1.254
    subnet: 255.255.255.0
```

### Anything in the logs that might be useful for us?

```txt
[I][web_server_base:046]: OTA Update Start: test-ota-factory.bin
[159940][D][Updater.cpp:133] begin(): OTA Partition: app1
[D][web_server_base:079]: OTA in progress: 0.1%
[D][web_server_base:079]: OTA in progress: 15.8%
[D][web_server_base:079]: OTA in progress: 34.4%
[D][web_server_base:079]: OTA in progress: 53.6%
[D][web_server_base:079]: OTA in progress: 69.3%
[D][web_server_base:079]: OTA in progress: 87.4%
[I][web_server_base:088]: OTA update successful!
[I][app:134]: Rebooting safely...
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[V][esp32.preferences:126]: sync: key: 233825507, len: 4
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

ssieb, 2023-08-25:
That's the wrong file.  The factory one is for serial flashing.

jmr61, 2023-08-26:
Thank you very much, I'm ashamed 👍

closed 2023-08-26 by jmr61