esphome/issues #4834: IPv6 cannot be turned off using ethernet

optimusprimespace, 2023-08-26:
### The problem

When i set the ipv6 to false i still get an ipv6 address that i can ping with another computer. Is this a bug or a feature? :) Also i'm using ethernet not wifi, as written in the "Example YAML snippet".

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.4

### What platform are you using?

ESP32

### Board

olimex gateway

### Component causing the issue

ipv6

### Example YAML snippet

```yaml
network:
  enable_ipv6: false
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
```

HeMan, 2023-08-28:
Seems that ESP-IDF (that Arduino for ESP32 also is built upon) defaults to enabling IPv6 with link local address, and it has been like that for a few years. I'll submit a PR that disables SLAAC.

HeMan, 2023-08-28:
> Seems that ESP-IDF (that Arduino for ESP32 also is built upon) defaults to enabling IPv6 with link local address, and it has been like that for a few years. I'll submit a PR that disables SLAAC.

I was wrong in my analyse, but the PR seem to be working.

optimusprimespace, 2023-08-29:
I can't wait for it to work. Thanks!