esphome/issues #4836: No documentation for `interval` component

clydebarrow, 2023-08-27:
### The problem

There is an `interval` component mentioned a few times in examples, but no documentation on it that I can find.

### Which version of ESPHome has the issue?

2023.9.0-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

interval

### Example YAML snippet

```yaml
interval :
  - interval: 2min
    then: ...
```

ssieb, 2023-08-28:
https://esphome.io/guides/automations#interval-component
https://esphome.io/search.html?q=interval

closed 2023-08-28 by ssieb

clydebarrow, 2023-08-28:
Ok, thanks. That does highlight why https://github.com/esphome/feature-requests/issues/2359 is important. Searching for "interval component" does not generate useful responses.