esphome/issues #4838: bluetooth_proxy not sending device "name" with advertisements

agittins, 2023-08-28:
### The problem

I have a development system with Home assistant and two esp32 bluetooth_proxy devices, but no local bluetooth adaptor. In the same building I have my production home assistant which has a local hci bluetooth usb adaptor, and a handful of esp32 bluetooth_proxy devices.

Of the 30-odd bluetooth MAC addresses in/near my house(!) only one displays a name on my development system.

On my production box, I get quite a few names (mostly my Xiaomi thermometers running pvvx firmware, some sending custom adverts, some sending BTHome v1).

Looking at the diagnostics for the bluetooth integration I can see that the only device name that is supplied by any of my esphome proxies is the `Pinetime` watch, and the only source for the other device names is the `hci0` adaptor on my production machine.

Some screenshots browsing the diagnostics - if the full file is required I can upload it but should probably wash all the macs from it first...

Discoveries from my `hubcentral` esphome bluetooth proxy, showing only `PineTime` with a name:
[image]

Same file, but showing the entries from the local usb bluetooth adaptor:
[image]

The interesting thing about the `PineTime` advertisement is that it looks like maybe it sends the `0x09` "Complete local name" GAP first in its packet, rather than at the end. The other devices appear to send their parts in numerical order of the type_id - some screenshots from NRFConnect on my phone:

[image]
(above device using pvvx custom ad format has no 0x01 flags block, but 0x16 followed by 0x09 local name).

[image]
(above uses pvvx BTHome v1 format, has flags, payload and local_name in that order)

[image]
(above is a cheap USB voltage tester, with 0x01, 0x03 and 0x09 GAP data blocks)

[image]
Above is the only device that gets a name from my proxies, note that it has the 0x09 local name GAP block first in the advertising packet.

If I enable debug logging for the esphome integration in homeassistant, I see entries like this: (note the absence of "name" fields, which I think is a change from the previous non-batched-non-raw adverts api)
[image]

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP32-IDF

### Board

TTGO-DISPLAY and D1-Mini32 clones

### Component causing the issue

bluetooth_proxy

### Example YAML snippet

```yaml
substitutions:
  devicename: prox-dining

esphome:
  name: $devicename

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

logger:
  baud_rate: 0 # disable serial uart logging

api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

switch:
  - platform: restart
    name: Reboot ${devicename}

esp32_ble_tracker:
  scan_parameters:
    active: false
    interval: 1000ms # suggested 211ms # default 320ms
    window:   900ms # suggested 120ms # default 30ms

bluetooth_proxy:
  active: False

sensor:
  - platform: wifi_signal
    name: "$devicename WiFi Signal"
    update_interval: 60s
```

### Additional information

I suspect that the new batched advertisement process should be parsing the name and including it in the advertisement protobuf alongside address and rssi - OR it should be sending advertisement packets containing the 0x09 portion of the advert.

(It's entirely possible I'm misreading anything in here, so feel free to question!)

bdraco, 2023-10-01:
Can you post the raw advertisements instead of images?

bdraco, 2023-10-01:
It looks like we have a parser for it 

we can add a test for the raw data though in case there is a bug
https://github.com/Bluetooth-Devices/bluetooth-data-tools/blob/main/src/bluetooth_data_tools/gap.py

bdraco, 2023-10-01:
Oh, never-mind you have `active: false` so it will never ask for the name

If you change `active: true` and recompile you will get names

agittins, 2023-10-01:
Ahh! Ok, that makes sense. I turned it back on and I am getting names showing up in the HA adverts dicts, thanks!

I guess that both NRFConnect on my phone and bluez on my main machine must be sending the scan-request packet which is why names would show up in their info.

(for anyone playing along at home, note it's the `esp32_ble_tracker`'s 'active` setting, not the `bluetooth_proxy`'s `active` setting).

closed 2023-10-01 by agittins