esphome/issues #4840: Issue with esphome + mqtt in Home Assistant 2023.8.x

legalgig, 2023-08-30:
### The problem

I'm using esphome devices with home assistant through mqtt for a long time now. With home assistant v2023.8.x there were some changes to mqtt naming scheme which left me with two warnings:
- Discovered entities with a name that starts with the device name
- Discovered MQTT entities with a name that is equal to the device name

so I've figured out I have to remove device name from esphome yaml files like this:

old yaml:
```
binary_sensor:
  - platform: status
    name: ${device_name} status

button:
  - platform: restart
    name: ${device_name} restart
```

new yaml:
```
binary_sensor:
  - platform: status
    name: status

button:
  - platform: restart
    name: restart
```

after changing the yaml and reflashing devices (I've also deleted old mqtt topics and device itself from home assistant) only one of them (entity had correct name like binary_sensor.<device_name>_restart) had those entities, the rest of them didn't show up. 

Home Assistant errored out with:
```
ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID ESPbinary_sensorstatus already exists - ignoring binary_sensor.some_sensor2_status
ERROR (MainThread) [homeassistant.components.button] Platform mqtt does not generate unique IDs. ID ESPbuttonrestart already exists - ignoring button.some_sensor2_restart
```

This issue was easily fixable (but required manually removing old entities) by setting mqtt unique id generator to mac
```
mqtt:
  broker: XXX.XXX.XXX.XXX
  topic_prefix: ${mqtt_prefix}/${device_name}
  discovery: true
  client_id: ${device_name}
  keepalive: 15s
  discovery_unique_id_generator: mac
``` 

But there is still one more issue with devices which should have entities named after device i.e.
```
light:
  - id: light_0
    name: ${device_name}
    platform: binary
    output: output_0
    restore_mode: RESTORE_DEFAULT_OFF
```

when I remove the name parameter then the light becomes internal (not intractable for the user). I've also tried some other ways, here is the list:

1. When the name is set to null
```
light:
  - id: light_0
    name: null
    platform: binary
    output: output_0
    restore_mode: RESTORE_DEFAULT_OFF
```
```
Failed config
light.binary: [source devices/../base/light_shelly1.yaml:20]
  id: light_0
  
  Expected boolean value, but cannot convert None to a boolean. Please use 'true' or 'false'.
  name: 
  platform: binary
  output: output_0
  restore_mode: RESTORE_DEFAULT_OFF
Invalid config for dummy-light!
```

2. When the name is set to false
```
light:
  - id: light_0
    name: false
    platform: binary
    output: output_0
    restore_mode: RESTORE_DEFAULT_OFF
```
```
Failed config
light.binary: [source devices/../base/light_shelly1.yaml:20]
  id: light_0
  
  Name cannot be None when esphome->friendly_name is not set!.
  name: False
  platform: binary
  output: output_0
  restore_mode: RESTORE_DEFAULT_OFF
Invalid config for dummy-light!
```

3. When the name is set to false and esphome.friendly_name to device_name it compiles but home assistant still complains about the naming scheme
```
light:
  - id: light_0
    name: false
    platform: binary
    output: output_0
    restore_mode: RESTORE_DEFAULT_OFF
```

4. when I do not set the name and set internal to false esphome just overwrites it with internal: true
5. I've also tried to set the name to null and none as a string but it just named the entity like this: dummy_light_none

Is this issue fixable by the user? 

P.S. This is a continuation of https://github.com/home-assistant/core/issues/98867

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.8.x

### What platform are you using?

ESP8266

raineth, 2023-09-05:
I've just started updating my configs as well and ran into the same problem.  By default, the unique IDs generated by the MQTT integration are "ESP&lt;component_type&gt;&lt;name&gt;" and relied on the entity names being made unique by the user (usually by prefixing a unique device name).

I worked around this by setting `discovery_unique_id_generator: mac` in my [mqtt configuration](https://esphome.io/components/mqtt#configuration-variables), which uses "&lt;mac&gt;&lt;component_type&gt;&lt;hash(name)&gt;", but IMO the legacy device ID generator should by changed to include the device's `friendly_name` when set due to the changes of [esphome#5205](https://github.com/esphome/esphome/pull/5205).

closed 2024-01-25 by github-actions[bot]

AndreKR, 2024-01-31:
I just ran into this as well. I have several Shelly i4 which all have a "Button 1". Their discovery `uniq_id´ all ended up being `ESPbinary_sensorbutton_1`.

Not including any device-specific identifier in the unique ID doesn't make sense to me. I think there are two possible solutions:
* Make `discovery_unique_id_generator: mac` the default.
* When it's left on the default `discovery_unique_id_generator: legacy`, include the node name in the `uniq_id`.