esphome/issues #4841: [bluetooth_proxy:439]: Only one API subscription is allowed at a time

drthanwho, 2023-08-30:
### The problem

Since I while my bluetooth sensors stopped comming through the esp bluetooth proxies to HA. On the other hand, on another HA that has bluetooth itself I can see the sensors comming through there so I know they are working. 

I hadn't found and errrors on the esp logs but just today I found the one bellow.

```[bluetooth_proxy:439]: Only one API subscription is allowed at a time```

### Which version of ESPHome has the issue?

2023.8.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Component causing the issue

bluetooth_proxy

### Example YAML snippet

```yaml
esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

packages:
  wifi: !include common_substitutions/wifi.yaml
  device_base: !include common_substitutions/device_base.yaml

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: false

bluetooth_proxy:
  active: false
```

rikroe, 2023-10-21:
I've seen the same when I tried debugging why one of my TRVs is not available anymore.
However I do not know if it is the culprit or connected in any way.

I'm using active bluetooth proxy and `esp32_ble_tracker` with default settings.

danswett, 2024-08-23:
I'm having this same issue and it's been going on for several months. I have three ESP32s set up as bluetooth proxies, and they'll work fine for a few days and then all of a sudden all my proxied bluetooth devices become unavailable. When I go pull logs from one of the ESPs they always have the error referred to in this ticket.

Here's the config from one of the ESPs:
esphome:
  name: pool-esp32
  friendly_name: Pool ESP32

esp32:
  board: esp32dev
  framework:
    type: esp-idf

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: redacted

ota:
  platform: esphome
  password: redacted

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

I don't know what causes it to work again - rebooting the ESPs sometimes seems to fix it.

ssieb, 2024-08-24:
You can't have 2 HA instances using the proxy at the same time.  That sounds like the issue for the initial post at least.

evlo, 2024-10-27:
can his be changed to have multiple HA connected to one bt proxy? With switch etc. it is not an issue

evlo, 2024-10-27:
workarounds are:
- ble to mqtt
- [ble client ](https://esphome.io/components/ble_client.html)for each device
- dedicated ble proxy for each instance

so far I'm think most reliable and sustainable would be ble client