esphome/issues #4848: empty ('') component name being replaced with device_name

Masterz69, 2023-09-02:
### The problem

While using esphome 2023.2.4 following config
```yaml
esphome:
  name: xxx
  friendly_name: yyy

switch:
  - platform: output
    name: ''
```
cause following entity_id in HA created: ```switch.yyy```

That is perfect for me naming - I like it very much. 

Below issue first time noticed in esphome 2023.6.2 and still in 2023.8.2. 
The same config now create entity_id ```switch.yyy_yyy``` created in HA - with duplicate device_name .

IMHO this is a bug.

Tried to find where it happens - in HA integration or in esphome software on device.
As demonstrated in log below - empty ('') component name now replaced with device_name on device side.

### Which version of ESPHome has the issue?

2023.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.4

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: xxx
  friendly_name: yyy

switch:
  - platform: output
    name: ''
```

### Anything in the logs that might be useful for us?

```txt
[23:14:34][C][output.switch:068]: Output Switch 'yyy'
```

droscy, 2023-09-14:
Try setting `name: None`.

From the [doc](https://esphome.io/components/switch/):

> If you have a friendly_name set for your device and you want the switch to use that name, you can set `name: None`.

Masterz69, 2023-09-14:
> Try setting `name: None`.
> 
> From the [doc](https://esphome.io/components/switch/):
> 
> > If you have a friendly_name set for your device and you want the switch to use that name, you can set `name: None`.

Sure, but... f.e. lets imagine following config:
```yaml
esphome:
  - friendly_name: ZZ

substitutions:
  prefix: None

switch:
  - platform: output
    name: ${prefix}

  - platform: template
    name: '${prefix} Inching'
```

In that i will NOT get entities ```switch.zz``` & ```switch.zz_inching```...
But ```switch.zz``` & ```switch.zz_none_inching```. 
Cause None is a just string in subsequent substitution parsing.

IMHO ```''``` must have same effect as ```None```. 
AS IT WAS in 2023 February f.e....

Weissnix4711, 2023-10-22:
This is happening to me with `name: None` as well. Seems like #4269 is making another appearance.

Masterz69, 2023-12-26:
During device configuration Validation in both cases ```name: None``` and ```name: ''``` see same output:

```yaml
switch:
- platform: output
  output: out_relay_1
  name: ''
```

But when firmware compiled and installed: ```name: ''``` appears as device name thus leading to duplicate device name in entity name in HA.

Masterz69, 2024-01-15:
[@unfocused](https://github.com/Unfocused), thanks for clear explanation of how works !include.

Can I ask You to check this topic and explain why when using device friendly name:
* `'None'` is replaced with empty component name;
* `''` is replaced with device friendly name... not with empty component name.

While `Validate` demonstrate '' as '' (empty as empty), actual compilation leads to HA enitity having device friendly name twice.

Masterz69, 2024-02-28:
ESPHome version 2023.12.7.

Jusr created `climate` as following:
```yaml
esphome:
  name: zzz-esp8285
  comment: ESP8285
  friendly_name: zzz ESP8285

climate:
  - platform: pid
    name: None
    id: climate_pid
```

As a results HA entity have name: `climate.zzz_esp8285_zzz_esp8285`.

So, declared in documentation NOT working IMHO.
`If you have a [friendly_name](https://esphome.io/components/esphome#esphome-configuration-variables) set for your device and you want the climate to use that name, you can set name: None.`

ssieb, 2024-02-28:
@Masterz69 that note in the doc is for the **name**, not the **entity id**.

Masterz69, 2024-02-28:
> @Masterz69 that note in the doc is for the **name**, not the **entity id**.

Note is about ESPHome component `name`, which is used to construct `entity_id` in HA.

Allow entity names to be set to None if the esphome -> friendly_name is set. 
This will then set the entity name to be blank and in Home Assistant this is handled by it being only give the device name.
^^ This was implemented by https://github.com/esphome/esphome/pull/4607, but looks not working in 2023.12.7.

ssieb, 2024-02-28:
Entity **name**, not **id**.  The name is set correctly, but how it maps to the id has nothing to do with that PR.

loryanstrant, 2024-03-31:
Seeing this as well. I have the following config:
[image]

What I end up with is:
[image]

jesserockz, 2024-03-31:
Can you please upload the [Device diagnostics](https://www.home-assistant.io/integrations/diagnostics/) for these devices added in HA.

This seems like it might be a bug on the HA side when it is generating the entity_id.

loryanstrant, 2024-04-01:
Sure thing, here you are!
[config_entry-esphome-eb73bc9e959ce5621233b55924bf26ec.json](https://github.com/esphome/issues/files/14819661/config_entry-esphome-eb73bc9e959ce5621233b55924bf26ec.json)

AJolly, 2025-02-22:
any updates on this?