esphome/issues #4850: esp-idf framework does not work, device enters boot loop

anton-malakhov, 2023-09-03:
### The problem

Created new empty device and changed framework type to `esp-idf`. It compiles and flashes successfully but does not work, staying in boot loop.

### Which version of ESPHome has the issue?

 2023.8.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

esp32-s3 devkitc-1 (YD-ESP32-S3 2022-V1.3)

### Component causing the issue

esp-idf

### Example YAML snippet

```yaml
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    #version: recommended
```

### Anything in the logs that might be useful for us?

```txt
QIO, clock div:1
load:0Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3808,len:0x162c
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Save
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3808,len:0x162c
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:0x3fce3808,len:0x162c
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
```

HeMan, 2023-09-03:
I've had similar problems when I got `board` wrong. Does it work with the arduino framework?

anton-malakhov, 2023-09-04:
> I've had similar problems when I got `board` wrong. Does it work with the arduino framework?

Yes, it works well, updated description. The `board` was detected automatically and looks correct

anton-malakhov, 2023-09-04:
This fixed the issue!
```
esphome:
  #...
  platformio_options:
    board_build.flash_mode: dio
```

closed 2023-09-04 by anton-malakhov