esphome/issues #4856: Ledc not compatible with libretiny anymore

hunhejj, 2023-09-06:
### The problem

First of all I am really happy, that finally Libretiny was merged into the official ESPHome. Great thanks to @kuba2k2 !

I wanted to migrate my existing configuration for a light, but got the following error:

> Component output.ledc requires component esp32.

### Which version of ESPHome has the issue?

2023.9.0-dev20230906

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

bk72xx

### Example YAML snippet

```yaml
esphome:
  name: smart-tuya-bulb

bk72xx:
  board: generic-bk7231n-qfn32-tuya

output:
  - platform: ledc
    id: output_red
    pin: P8
```

kuba2k2, 2023-09-06:
LibreTiny now uses libretiny_pwm instead of ledc. Simply change that in your yaml and it will work.

hunhejj, 2023-09-06:
Thanks for the info @kuba2k2 !

closed 2023-09-06 by hunhejj