esphome/issues #4859: Can't install firmware for libretiny on Raspberry

hunhejj, 2023-09-06:
### The problem

I created a configuration for libretiny and tried to compile it on a 64bit Raspberry. I got the following error:

> UnknownPackageError: Could not find the package with 'libretiny @ 1.3.0' requirements for your system 'linux_aarch64'

### Which version of ESPHome has the issue?

2023.9.0-dev20230906

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

hunhejj, 2023-09-06:
@kuba2k2

kuba2k2, 2023-09-06:
Publishing to PlatformIO has been delayed for an unknown reason. Please use this for now:
```yaml
bk72xx:
  framework:
    version: dev
```

hunhejj, 2023-09-06:
Sorry to hear that. Well I will stick to the `dev` version then. As a reference for others here‘s the related thread: https://community.platformio.org/t/libretiny-platform-version-publish-delay

closed 2023-09-06 by hunhejj