esphome/issues #4860: Global Bool Variables not working

rmandjes, 2023-09-06:
### The problem

I'm making use of Global Bool variables where I want to enable some starting values.
However, it doensn't matter what the value is of this Bool variable, the result is that value is always False.

It' has worked in an older version of ESPhome, but with the newer versions it goes wrong.

Here is a part of the code I'm using:
globals:
  - id: daytime_start_hour
    type: int
    restore_value: no
    initial_value: "8"
  - id: daytime_end_hour
    type: int
    restore_value: no
    initial_value: "22"
  - id: show_good_status # if false: show only on orange and red not green
    type: bool
    restore_value: true
    initial_value: "true"
  - id: night_mode_enabled # if true: turn off all leds during from 22 to 8h
    type: bool
    restore_value: true
    initial_value: "true"
  
...

esp8266:
  board: d1_mini

So when I start my board, the show_good variable has the value False, instead of True.

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3 and 2023.8.2

### What platform are you using?

ESP8266

### Board

d1 Mini

### Component causing the issue

Global Variables

Lumpusz, 2023-09-06:
Is thi by any chance a duplicate of this issue: https://github.com/esphome/issues/issues/4796

ssieb, 2023-09-09:
Duplicate of #4796

closed 2023-09-09 by ssieb

rmandjes, 2023-09-09:
See my comment by 4796. This option does not exists voor global variables.

ssieb, 2023-09-09:
It has nothing to do with the global.  It's the switch that's the problem.  If you have more questions about this, come ask on the esphome discord server.