esphome/issues #4861: After update to 2023.8.3 device is not accessible

daha76, 2023-09-06:
### The problem

Hello,

Today I made update from 2023.8.2 to 2023.8.3 of esphome and firmware of my ESP8266 device. From that time ESPHome detects my decide as online, but I can't get logs or any data from it. ESPHome running as addon of Home assistant on RPi4. 
When I try to get logs this error pops up:

WARNING Can't connect to ESPHome API for heater.hdhome.sk: Error connecting to ('192.168.2.2', 6053): [Errno 111] Connect call failed ('192.168.2.2', 6053) (SocketAPIError)
I tried to restore previous version but it didn't help.

Device is not easy reachable, 200km away, so I don't know if I can repair. Any ideas?

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.4

### What platform are you using?

ESP8266

### Board

WemoD! Mini

### Component causing the issue

Network

### Anything in the logs that might be useful for us?

```txt
WARNING Can't connect to ESPHome API for heater.hdhome.sk: Error connecting to ('192.168.2.2', 6053): [Errno 111] Connect call failed ('192.168.2.2', 6053) (SocketAPIError)
```

daha76, 2023-09-06:
After resetting both routers, VPN, and Rpi several times it started to work as before. I have no idea what happened.

closed 2023-09-06 by daha76

t34wrj, 2023-09-06:
I also have this issue.