esphome/issues #4862: seeed_xiao_esp32c3 Board Unavailable until connecting to port

flixlix, 2023-09-06:
### The problem

After uploading new firmware to my seeed_xiao_esp32c3 board, I noticed that is stays offline until I connect to its USB port and open the logs in es-home through USB. If I don't connect to the Logs it doesn't ever get online but also doesn't use the fallback hotspot...

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.1

### What platform are you using?

ESP32

### Board

ESP32-c3

### Example YAML snippet

```yaml
esphome:
  name: meter-water
  friendly_name: meter-water
  includes:
    - my_custom_component.h

esp32:
  board: seeed_xiao_esp32c3
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Fs2wo4KFz8RZtfULDX/hGr3W2sA/vASOqBuHWl+uDss="

ota:
  password: "ef7472a1da570a3be2f215c9dbafa795"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

    
web_server:
  port: 80

sensor:
  - platform: custom
    lambda: |-
      auto my_custom = new MyCustomSensor();
      App.register_component(my_custom);
      return {my_custom->water_liter_sensor, my_custom->sensa, my_custom->sensb, my_custom->sensc};

    sensors:
      - name: "water_liter_sensor"
        unit_of_measurement: L
        state_class: total_increasing
        device_class: water
        accuracy_decimals: 2
      - name: "sensa"
        accuracy_decimals: 2
      - name: "sensb"
        accuracy_decimals: 2
      - name: "sensc"
        accuracy_decimals: 2
```

### Anything in the logs that might be useful for us?

```txt
After opening the logs, the first thing showing up is this:

There have been 3 suspected unsuccessful boot attempts.
```
```

c-MM, 2023-09-06:
Do you know what your last working version was?
Maybe it is related to my problem in issue #4864  not booting with  external power.

martijnvwezel, 2023-09-10:
@c-MM  This board that he uses is powered by usb-c. This should just work, with custom firmware or very old firmware it keep working..

ssieb, 2023-12-14:
Try adding `baud_rate: 0` to the logger.

frroberts, 2023-12-15:
I can confirm that setting `baud_rate: 0` in the logger section seems to fix the issue and the device boots even with just power to the USB port

martijnvwezel, 2023-12-16:
@ssieb  and @frroberts  I fixed it in my yaml in my repository: [githubpage](https://github.com/martijnvwezel/watermeter-esphome/blob/main/muino-water-meter-esp32.yaml), basically the  line `hardware_uart: USB_SERIAL_JTAG`, is the one you should select

ssieb, 2023-12-16:
Yes, but that only works if you're using esp-idf, not arduino, unless you're using dev or beta.

martijnvwezel, 2023-12-16:
ahh yes, you are right! I solved it at that time by setting to a different uart