esphome/issues #4864: Mini D1 ESP32 doesn't boot with power on VCC pin after upgrade to ESPHOME 2023.8.1

c-MM, 2023-09-06:
### The problem

After upgrading my D1 mini ESP32 boards (clones from AZdelivery on Amazon) to ESPHOME 2023.8.{1,2,3} they don't boot anymore when powered via the VCC and GND pins. They still work fine when powered via USB.

Flashing the boards via USB back to ESPHOME 2023.7.1 makes them work again when powered via VCC and GND pins.

I didn't try ESPHOME 2023.8.0 but the issue exists with 2023.8.1, 2023.8.2 and 2023.8.3.

### Which version of ESPHome has the issue?

2023.8.1, 2023.8.2 and 2023.8.3 (I suspect  2023.8.0 too)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

ESP32

### Board

D1 mini ESP32 from AZdelivery (clone of Wemos D1 mini ESP32)

### Component causing the issue

core, boot

### Example YAML snippet

```yaml
substitutions:
  name: btp1
esphome:
  name: ${name}
  name_add_mac_suffix: false
esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "eWVpMUZlZTlUaGllMnF1b282dUNoNGthaE1vaGNpaTc="

ota:
  password: "<password>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

improv_serial:

esp32_ble_tracker:
  scan_parameters:
    window: 1100ms
    interval: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

TurboMOD, 2023-10-07:
Any news on this? It looks like iam facing the same issue. Iam on Current version: 2023.9.3 and my D1 mini ESP32 just dont get online in esphome with powering via VCC pin (5V).  If I use the USB port, everything works fine. Unfortunately I already made a PCB including a 5V regulator to power the ESP and some other parts.

c-MM, 2023-10-14:
Same Problem 2023.9.3 here, I stick to 2023.7.1 for now. I guess some change in the initialisation keeps the ESP from booting, when the USB-serial converter is not powered.

asitwouldseem, 2024-01-09:
Same issue here running 2023.12.5 on a D1 mini clone via the 5V pin.

ssieb, 2024-01-10:
Can you get serial logs?  Use a USB TTL uart adapter but only connect GND and RX pins on it.

c-MM, 2024-01-14:
Ok, I did a bit of soldering. 
The log says:
[...]
[D][esp-idf:000]: I (1341) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07

Brownout detector was triggered

Was there a change concerning the brownout detector between 2023.7.1 and 2023.8.1?
The 2023.7.1 version still works fine on the board.

Is there a way to disable brownout in the config yaml?

c-MM, 2024-01-14:
I disabled CONFIG_ESP32_BROWNOUT_DET in sdkconfig and it doesn't help. The module just crashes. It also doesn't matter which power supply I use. So there might be some new IO-config starting with 2023.8.x that drains too much power when USB power is not available.

c-MM, 2024-01-15:
I suspected something like this. But the question remains, why it works with 2023.7.1 an lower. 
There must be some changes in the gpio setup leading to the crash/power use.