esphome/issues #4870: ESP32-C3 Doesn't boot/connect after reset/repowering

martijnvwezel, 2023-09-10:
### The problem

If you install the ESPHome on a ESP32C3 it works. However after a reboot/repower/restart if fails to work. Its hard to debug, because the moment you debug/open serial connection it works and keeps on working. If you update/install and keep it plugged it keeps working. After it lost its WiFi it gets a reboot and stops from working.

### Which version of ESPHome has the issue?

ESPHome 2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest and before

### What platform are you using?

ESP32

### Board

Muino Water Meter

### Example YAML snippet

```yaml
substitutions:
  name: muino-water-meter-ddb02c
packages:
  esphome.muino-water-meter: github://martijnvwezel/watermeter-esphome/muino-water-meter-esp32.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: redacted-key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
Cannot log, otherwise it works..
```

### Additional information

This project: https://watermeter.muino.nl/

rob-rocks, 2024-01-06:
I'm having the same problem with an ESP32-C3 Super Mini [ESP32-C3 Super Mini](https://homeding.github.io/boards/esp32c3/super-mini-c3.htm) board. I thought it might be related to [Boot loop on ESP32-C3](https://github.com/platformio/platform-espressif32/issues/622) and I tried the 
`board_build.flash_mode: dio`
setting, but so far no improvement. Have you been able to find a solution?

MacksMaster, 2024-01-13:
Same here with this board [https://www.aliexpress.com/item/1005006165377284.html?spm=a2g0o.order_list.order_list_main.64.502c1802X4Kokh](url)

Tendonitis, 2024-02-11:
I'm having the same issue with my ESP32-C3 Super Mini connected to an LD2410C.

martijnvwezel, 2024-02-11:
See https://github.com/martijnvwezel/watermeter-esphome
look at platform io thingy and the logger

martijnvwezel, 2024-02-11:
BAsically the flash type is wrongly selected

closed 2024-02-11 by martijnvwezel