esphome/issues #4872: ESPHome st7789v LCD Display No Longer Operational After 2023-7 or 2023-8 updates

TDStrasser, 2023-09-11:
### The problem

Somewhere in the last 6 months a firmware change has rendered the st7769v display, used with a RP2040 Pico-W processor no longer functional. It was working on my installation (yaml attached) until the firmware was updated from about 6 months ago to the latest 2023-8-3 release.
[esphome-hydroponics.zip](https://github.com/esphome/issues/files/12573307/esphome-hydroponics.zip)

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.8.3

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico-W

### Component causing the issue

st7789v

### Example YAML snippet

```yaml
esphome:
  name: esphome-hydroponics
  friendly_name: hydroponics

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git
# Colour LCD Display
spi:
  clk_pin: 6
  mosi_pin: 7
  miso_pin: 4
  id: spi_0

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_grey
    red: 70%
    green: 70%
    blue: 70%
  - id: my_white
    red: 100%
    green: 100%
    blue: 100%
  - id: my_cyan
    red: 0%
    green: 100%
    blue: 100%
  - id: my_purple
    red: 100%
    green: 0%
    blue: 100%

font:
  - file: "Helvetica.ttf"
    id: helvetica_48
    size: 48
  - file: "Helvetica.ttf"
    id: helvetica_36
    size: 36
  - file: "Helvetica.ttf"
    id: helvetica_24
    size: 24
  - file: "Helvetica.ttf"
    id: helvetica_18
    size: 18
  - file: "Helvetica.ttf"
    id: helvetica_S
    size: 12
# put a ttf font called Helvetica.ttf in the esphome folder

display:
  - platform: st7789v
    model: custom
    height: 320
    width: 240
    offset_height: 0
    offset_width: 0
    eightbitcolor: True
    backlight_pin: 12
    cs_pin: 5
    dc_pin: 13
    reset_pin: 17
    rotation: 0
    lambda: !lambda |-
      // Controller Status
      it.print(0, 0, id(helvetica_24), id(my_yellow), "Controls");

      if(id(wifi_status)) {
        it.print(140,0, id(helvetica_18), id(my_green), "WiFi OK");
      }
      else {
        it.print(140,0, id(helvetica_18), id(my_red), "No WiFi");
      }

      // Hydroponics Status
      it.print(0, 70, id(helvetica_24), id(my_yellow), "Hydroponics");

      it.print(0, 100, id(helvetica_18), id(my_white), "Tank:");
      if(id(hydroponics_water_level).state) {
        it.print(60, 100, id(helvetica_18), id(my_green), "OK");
      }
      else {
        it.print(60, 100, id(helvetica_18), id(my_red), "Low");
      }

      it.print(110, 100, id(helvetica_18), id(my_white), "Pump:");
      if(id(hydroponics_pump).state) {
        it.print(170, 100, id(helvetica_18), id(my_green), "On");
      }
      else {
        it.print(170, 100, id(helvetica_18), id(my_red), "Off");
      }

      it.print(0, 120, id(helvetica_18), id(my_white), "Flow:");
      if(id(hydroponic_water_flow).state > 1) {
        it.printf(80, 120, id(helvetica_18), id(my_green), "%.1fL/min OK", id(hydroponic_water_flow).state);
      }
      else {
        it.printf(80, 120, id(helvetica_18), id(my_red), "%.1fL/min Low", id(hydroponic_water_flow).state);
      }

      it.print(0, 140, id(helvetica_18), id(my_white), "Salts:");
      if(id(tds_sensor).state > 600) {
        it.printf(80, 140, id(helvetica_18), id(my_green), "%.0fppm OK", id(tds_sensor).state);
      }
      else {
        it.printf(80, 140, id(helvetica_18), id(my_red), "%.0fppm Low", id(tds_sensor).state);
      }

      it.print(0, 160, id(helvetica_18), id(my_white), "Water Temp:");
      it.printf(120, 160, id(helvetica_18), id(my_green), "%.1fC", id(water_temperature).state);

      // Vegetable Pod Status
      it.print(0, 200, id(helvetica_24), id(my_yellow), "Veggie Pod");

      it.print(0, 230, id(helvetica_18), id(my_white), "Tank:");
      if(id(veggie_box_water_level).state) {
        it.print(60, 230, id(helvetica_18), id(my_green), "OK");
      }
      else {
        it.print(60, 230, id(helvetica_18), id(my_red), "Low");
      }

      it.print(110, 230, id(helvetica_18), id(my_white), "Pump:");
      if(id(spray_pump).state) {
        it.print(170, 230, id(helvetica_18), id(my_green), "On");
      }
      else {
        it.print(170, 230, id(helvetica_18), id(my_red), "Off");
      }

      it.print(0, 250, id(helvetica_18), id(my_white), "Soil Moisture:");
      if(id(soil_moisture_sensor).state > 50) {
        it.printf(140, 250, id(helvetica_18), id(my_green), "%.0f%% OK", id(soil_moisture_sensor).state);
      }
      else {
        it.printf(140, 250, id(helvetica_18), id(my_red), "%.0f%% Low", id(soil_moisture_sensor).state);
      }

      // Ambient Environment
      it.print(0, 280, id(helvetica_24), id(my_yellow), "Ambient");

      it.printf(120,280, id(helvetica_18), id(my_cyan), "%.1fC", id(ambient_temperature).raw_state);
      it.printf(180,280, id(helvetica_18), id(my_cyan), "%.0f%%", id(ambient_humidity).raw_state);
```

### Anything in the logs that might be useful for us?

```txt
Display driver issues diagnostic messages that the st7789v is slower than expected. This may indicate the SPI clock rate is slow as the display chip doesn't respond with replies to screen updates. e.g.
[00:00:00][W][component:204]: Component st7789v.display took a long time for an operation (0.79 s).
[00:00:00][W][component:205]: Components should block for at most 20-30ms.
Strongly suspect the changes in the update 2023.7.0 have affected the proper functioning of the display.
```

### Additional information

I can see the display backlight is on, so it is a lack of SPI updates to the display.
No other error messages apparent.
The Pico processor has enough RAM to handle the relatively large display as it was working prior to this update. I hadn't updated the Pico for the prior 6 months as I hadn't been working on it.
I got a workaround using the st7735 working after some experimentation.  This proves the SPI bus is working and the hardware is properly configured (and it was working before the last or 2023-7 firmware drops.
The hardware is a DFRobot 2.0" 320x240 IPS LCD (V1.0) which uses the ST7789V display chip.
[config_entry-esphome-e2a91e090869e864a0f553bca6ae991b.json.txt](https://github.com/esphome/issues/files/12573291/config_entry-esphome-e2a91e090869e864a0f553bca6ae991b.json.txt)

nielsnl68, 2023-09-11:
The warning was always there, the only reason you see it now is that Jesse has change the log level for this warning from Verbose to Warning. 

Is there anything else in the log that could tell why it does not work, maybe show the whole log?

nielsnl68, 2023-09-11:
Also do you have any ESP32 around that you can use to test this display with?

TDStrasser, 2023-09-11:
Nothing else is in the log relating to the display or SPI.  There is no failure indication available from the display hardware so only a visual confirmation that the display is working correctly is available. Nonetheless the console log is attached.
I don't have an ESP32 handy to try it. 
I am reporting an issue of a now apparently non-functioning element resulting from a firmware update.
I know the hardware is fine because the st7735 workaround does work.
[logs_esphome-hydroponics_run.txt](https://github.com/esphome/issues/files/12574689/logs_esphome-hydroponics_run.txt)

TDStrasser, 2023-09-30:
Update 2023.9.1 seems to have restored the 7789v, but now the SPI is very slow on both the st7789v and st7735 configurations.  This has been reported as a new issue. This one can be closed.

closed 2023-09-30 by TDStrasser