esphome/issues #4876: No IPv6 with ESP8266 even with enable_ipv6 set to true

Nardol, 2023-09-13:
### The problem

I've modified my ESPHome config to enable IPv6 for ESP8266.
I have no compilation error but even in very_verbose log level, I have nothing displayed about IPv6 at all.

### Which version of ESPHome has the issue?

2023.9.0b1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

network

### Example YAML snippet

```yaml
esphome:
  name: ballon-eau-chaude
  project:
    name: "Shelly.ballon_eau_chaude"
    version: "1.3.0"

esp8266:
  board: esp01_1m
  restore_from_flash: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key
  ap:
    password: !secret ap_key

network:
  enable_ipv6: true

captive_portal:

logger:
  baud_rate: 0
  level: ERROR # Same with very_verbose

api:
  encryption:
    key: !secret ballon_eau_chaude_encryption_key

ota:
  password: !secret ballon_eau_chaude_ota_pass

switch:
- platform: gpio
  name: "Ballon d'Eau Chaude"
  icon: "mdi:water-boiler"
  pin: GPIO4
  id: ballon

binary_sensor:
- platform: gpio
  pin:
    number: GPIO5
  name: "Switch Shelly 1"
  on_state:
    then:
    - switch.toggle: ballon
  internal: true
  id: switchid

time:
- platform: homeassistant
  id: my_timer
  timezone: Europe/Paris

button:
- platform: factory_reset
  name: Réinitialiser et redémarrer Ballon d'eau chaude
  entity_category: "diagnostic"
- platform: restart
  name: "Redémarrer Ballon d'eau chaude"
  entity_category: "diagnostic"
```

### Additional information

I can send a log in very_verbose if needed but when I read one, I had no reference to IPv6 at all.
Unfortunately I have no ESP8266 I can connect to directly using serial, only Wi-Fi.
And I have no other board to test with, all my ESP8266 are esp01_1m.

HeMan, 2023-09-13:
I will unfortunately only show which IPv6 address it got on the serial log. I'll try to find another way that you can see it.

Nardol, 2023-09-14:
@HeMan  To be sure, I confirm there is nothing logged but it is confirmed no IPv6 is set.
No ping possible using IPv6 and using dig only IPv4 is returned.
So it looks like it is not only a question of nothing logged but I have no IPv6 at all for my ESP8266 devices.
And it is the same in every log levels
I prefer giving this precision because when reading again the description I made, I think it can be ambiguous.
I also update this issue description to be sure.

HeMan, 2023-09-14:
> @HeMan To be sure, I confirm there is nothing logged but it is confirmed no IPv6 is set. No ping possible using IPv6 and using dig only IPv4 is returned. So it looks like it is not only a question [...]

Te be clear, it won't use DHCPv6 (just SLAAC) so dig won't be able to query your dns for IPv6 address, and mDNS does currently not work with IPv6.
Could you please attach a VERY_VERBOSE log?

HeMan, 2023-09-14:
Just tried your config and a VERY_VERBOSE won't show anything unfortunately. I'll try to add some way of getting IPv6 addresses in the log.

Nardol, 2023-09-14:
> Te be clear, it won't use DHCPv6 (just SLAAC) so dig won't be able to query your dns for IPv6 address, and mDNS does currently not work with IPv6. Could you please attach a VERY_VERBOSE log?

I use SLAAC.
With dig I type:
`dig ballon-eau-chaude.local +short @<IPv4_of_esp> -p5353`
So if you tel me MDNS does not work with IPv6, I now understand why this method and ping to the .local address don't work.

Finally I asked someone who gave me the IPv6 address from the MAC of the ESP8266, which pings right.
And [this website to convert MAC to IPv6](https://ben.akrin.com/mac-address-to-ipv6-link-local-address-online-converter/).

Things will be really simpler when MDNS will work with IPv6 with ESP8266 :slightly_smiling_face: anyway thanks for your work especially for IPv6.

Sorry for the noise.

closed 2023-09-14 by Nardol

HeMan, 2023-09-20:
I have a initial support for reporting all ip addresses both via debug and `wifi_sensor` if you are interested in testing, https://github.com/HeMan/esphome/tree/improve-dualstack
It is far from done, but could be interesting to test.

Nardol, 2023-09-21:
Thanks, interesting.
I'm going to test.
So I should use the `ip_address` text sensor which should display both IPv4 and IPv6 if I understand correctly?
Or there are also changes in the logger using Wi-Fi?

HeMan, 2023-09-21:
> Thanks, interesting. I'm going to test. So I should use the `ip_address` text sensor which should display both IPv4 and IPv6 if I understand correctly? Or there are also changes in the logger using Wi-Fi?

Yes, it's both the `ip_address` text sensor and the logger (and mqtt) that shows all the addresses it got.

jeremyp3, 2023-09-21:
I've just tested this branch on my ESP32 and it works. however, we have the same problem as in #3896, it only displays the IPV6 fe80 but not the IPV6 slaac, but this may be known and normal :)

HeMan, 2023-09-21:
If you like to test some more you could change idf and platformversion with:
```
    version: 5.0.2
    platform_version: 6.3.2
```

Nardol, 2023-09-21:
Should it work with an ESP8266?

In my case it does not compile with the following error:

```txt
src/esphome/components/captive_portal/captive_portal.cpp: In member function 'void esphome::captive_portal::CaptivePortal::start()':
src/esphome/components/captive_portal/captive_portal.cpp:51:49: error: call of overloaded 'IPAddress(esphome::network::IPAddress&)' is ambiguous
   51 |   this->dns_server_->start(53, "*", IPAddress(ip));
      |                                                 ^
In file included from /home/pzajda/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Udp.h:39,
                 from /home/pzajda/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiUdp.h:25,
                 from /home/pzajda/.platformio/packages/framework-arduinoespressif8266/libraries/DNSServer/src/DNSServer.h:3,
                 from src/esphome/components/captive_portal/captive_portal.h:5,
                 from src/esphome/components/captive_portal/captive_portal.cpp:1:
/home/pzajda/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/IPAddress.h:172:9: note: candidate: 'IPAddress::IPAddress(const ip_addr_t&)'
  172 |         IPAddress(const ip_addr_t& lwip_addr) { ip_addr_copy(_ip, lwip_addr); }
      |         ^~~~~~~~~
/home/pzajda/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/IPAddress.h:155:9: note: candidate: 'IPAddress::IPAddress(const ip4_addr&)'
  155 |         IPAddress(const ipv4_addr& fw_addr)   { setV4(); v4() = fw_addr.addr; }
      |         ^~~~~~~~~
```

Nardol, 2023-09-21:
Nice, it works. Thanks!
I also tested with a wifi_info ip_address text_sensor on both ESP32 IDF and ESP8266 which worked well.

jeremyp3, 2023-09-21:
upgrading to the above-mentioned version doesn't seem to solve the display bug as far as I'm concerned. I have an az-delivery-devkit-v4

it still only displays fe80 in v6.
curl -u xxx:yyy http://$hostname/text_sensor/esp_ip_address
{"id":"text_sensor-esp_ip_address","value":"IPv4: 192.168.80.29\nIPv6: FE80::3E61:5FF:FE31:4660\n","state":"IPv4: 192.168.80.29\nIPv6: FE80::3E61:5FF:FE31:4660\

HeMan, 2023-09-21:
We could continue the discussion in this Discord thread,  https://discord.com/channels/429907082951524364/1094669489561424023

jeremyp3, 2023-09-21:
I have to join the server first, don't I?

because when I arrive on discord, I can't do anything I can see the channel name "more ipv6".

my pseudo is the same

thanks