esphome/issues #4877: Trying to enforce connecting to a Mac address / repeater

swingstate, 2023-09-13:
### The problem

I am using a ESP32 board and have a Mesh WIFI running. I want the ESP to connect to a certain mac / repeater in that mesh, however it doesn't do that. 

I am trying to use the bssid option but that doesn't do it. Without the bssid option and with the standard: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

 the board instantly connects. 

Any advice?

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32U / ESP32-DevKitC V4.0

### Component causing the issue

wifi

### Example YAML snippet

```yaml
esphome:
  name: esp32--solar-sensorik-antenne
  friendly_name: ESP32 - Solar Sensorik plus Antenne 

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: ##################

ota:
  password: "##################"

wifi:
  fast_connect: true
  networks:
  - ssid: Skynet
    password: xxxxxxxxx
    bssid: ##:##:##:##:##:##
    channel: 6

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32--Solar-Sensorik-antenne"
    password: "##########"

captive_portal:

sensor:
  - platform: wifi_signal
    name: "WIFI Signal Quality"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[16:07:21][I][wifi:274]: WiFi Connecting to '#########'...
[16:07:24][W][wifi:598]: WiFi network can not be found anymore.
[16:07:24][W][wifi_esp32:504]: Event: Disconnected ssid='#######' reason='Probe Request Unsuccessful'
[16:07:24][W][wifi:624]: Restarting WiFi adapter...
[16:07:24][I][wifi:274]: WiFi Connecting to '#########'...
[16:07:26][W][wifi_esp32:504]: Event: Disconnected ssid='######' reason='Probe Request Unsuccessful'
[16:07:26][W][wifi:598]: WiFi network can not be found anymore.
[16:07:26][W][wifi:624]: Restarting WiFi adapter...
```

ssieb, 2023-09-14:
I just tested this and it works.  Are you sure you have the right bssid?  Maybe try without the fast connect and see what it can see.

swingstate, 2023-09-16:
Still testing - do I have to provide the SSID when I use the BSSID option?

ssieb, 2023-09-16:
Yes.

swingstate, 2023-09-16:
Thanks! Do I have to use plain text for SSID & Password or can I also use !secret wifi_ssid / !secret wifi_password.

I also did some testing yesterday and even with the BSSID being listed in the scan the ESP refused to connect.

ssieb, 2023-09-16:
You can always use plain text.  The `!secret` option is just to keep the plain text out of the yaml file if you want.

swingstate, 2023-09-20:
After couple days testing this seems to be stable, so I'll close the issue here.

closed 2023-09-20 by swingstate