esphome/issues #4881: [GUI] Cannot do wireless update (OTA) when using HA and SSL port other than 443

elproko, 2023-09-14:
### The problem

When using HA and SSL nginx reverse proxy addon which is set to port other than 443, ESPHome cannot update devices from dashboard.
The window where normally compile & OTA programming progress is shown now shows nothing.
Update never finishes.

The issue persists for individual devices (Install/Update-->Wirelessly) and for "Update All" menu item.

### Which version of ESPHome has the issue?

2023.8.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.1

### What platform are you using?

ESP8266

### Board

any

### Component causing the issue

gui

### Example YAML snippet

```yaml
no yaml involved
```

### Anything in the logs that might be useful for us?

```txt
no info in logs
```

elproko, 2023-09-14:
Example SSL URL when issue is **present**:
http**s**://ha.somedomain.tld:**444**/5c53de3b_esphome/ingress

Example SSL URL when issue does not exist:
http**s**://ha.somedomain.tld/5c53de3b_esphome/ingress

Example plain HTTP URL when issue does not exist: 
http://ha.somedomain.tld:8123/5c53de3b_esphome/ingress

ogghi, 2023-10-04:
Hi there, I am having the same issue!

Any solution would be nice :+1:

elproko, 2023-10-25:
> Hi there, I am having the same issue!
> 
> Any solution would be nice 👍

It seems connected to websocket issue when using other then 443 port.

xe5700, 2023-10-28:
I have same issue